bigtreetech/EBB #2: Max31865 Overvoltage or undervoltage fault

710920772, 2022-06-02:
Max31865 Overvoltage or undervoltage fault
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

When i was printting,It suddenly appeared!!!!!
I don't konw why????????????
the wire connected solid! And no indication towards Leakage!It just suddenly happen!!!!!!!!

Voodoobrew101, 2022-10-09:
If you power by a smart usb charger, the power in usb c is not fixed. I don't know what controlls it. But I suspect the charger is doing an assessment to decide what power to provide. Also faulty peripherals can cause or possibly housing power brownouts caused by say...a refrigerator compressor kicking on.

Get a proper pi power supply and the problem should go away. If you really never want to see it also get a battery backup with power conditioner.

se508, 2023-02-22:
I had the same issue and solved it.
It was on a Voron with Octopus Pro board. On the Voron the hotend is not grounded, so the potential is floating. With a heater connected even a very small leak current may be problematic.
One solution is to ground the hotend to 0v rail.
Another is to ensure there is no leak current from the heater to the sensor (replacing the heater in my case).