bigtreetech/EBB #3: Communication timeout during homing probe

710920772, 2022-06-02:
Communication timeout during homing probe

McPrince96, 2022-06-14:
Did you find a solution in the meantime? I ran into ai similar timeout but with x homing. 

Looking at when your issue was opened i don't believe BTT is very active here. *sigh*

murphytodd4, 2022-09-09:
I had the same issue and ended up increasing my CAN baud rate to 500000. I still have the error once in a while but it is far more stable. I was also not able to run an resonance test successfully before increasing the rate.

basha318i, 2022-09-17:
> I had the same issue and ended up increasing my CAN baud rate to 500000. I still have the error once in a while but it is far more stable. I was also not able to run an resonance test successfully before increasing the rate.

Hi, I have the same issue but it’s happening quite frequently! I’m at 500,000 baud right now & not able to complete a single z-tilt!
Which version do you have on the U2C? The one on BTT github?

murphytodd4, 2022-09-17:
> Hi, I have the same issue but it’s happening quite frequently! I’m at 500,000 baud right now & not able to complete a single z-tilt!
>
> Which version do you have on the U2C? The one on BTT github?

I believe I said that wrong. I set it to 500000 in the device tree setup but then ran my boards at 250000. I reflashed my firmware today at 500000 and the issue came back. 

I am also just running a generic CAN to SPI breakout board.

McPrince96, 2022-09-17:
I'm not really sure what the solution was for me but this is what i did:
- Complete fresh install
- Orange Pi 3 lts instead of rpi zero
- Other higher quality usb cables

Be sure to watch this [this video](https://youtu.be/_FELCN8CbWA) . The can software mentioned in the btt manual is crap and installation is a pain in the ass. The can software mentioned in the video didn't used to support the Gb0 chip but now it should support it.

Voodoobrew101, 2022-10-09:
I have this error when my bed is too out of level. Try doing a manual level first. 

Also give the z min travel -5. Sorry don't have exact parameter name on phone.

murphytodd4, 2022-10-09:
I ended up following this reply [here](https://klipper.discourse.group/t/canbus-communication-timeout-while-homing-z/3741/3) and I have not had another timeout error while trying to home

TURNERO, 2022-11-13:
I got this and it got worse over time. I was tearing my hair out. I kept getting undervoltage errors on my PI which was causing it to throttle. I was using a tablet plugged into one of the USB ports on the PI as a screen. Unplugging that seems to have fixed it.

emersonsc, 2023-02-06:
> I ended up following this reply [here](https://klipper.discourse.group/t/canbus-communication-timeout-while-homing-z/3741/3) and I have not had another timeout error while trying to home.
> 
> Basically changing TRSYNC_TIMEOUT value in “/home/pi/klipper/klippy/mcu.py” file from 0.025 to 0.050

Tried that, didnt help for me.

Voodoobrew101, 2023-02-07:
My problem was bad plug conection. If you wiggle it and the light goes
off...

LobodPl, 2023-03-25:
If you are using "dockable_probe.py" module. Take a look [here ](https://github.com/Klipper3d/klipper/pull/4328#issuecomment-1476612792).

Tom-T, 2023-04-15:
> I ended up following this reply [here](https://klipper.discourse.group/t/canbus-communication-timeout-while-homing-z/3741/3) and I have not had another timeout error while trying to home.
> 
> Basically changing TRSYNC_TIMEOUT value in “/home/pi/klipper/klippy/mcu.py” file from 0.025 to 0.050

This seems to have resolved my issue. Thank you for posting about it. :)

florianmuetzel, 2023-05-06:
I had the same today. But I am not using can. I Am using just the usb connection.  My problem was also a bad connection! A little bit of hot glue on the dupont plug resolved the problem