bigtreetech/EBB #12: I2C timeout error

DrumClock, 2022-08-12:
I connected an OLED display to the I2C connector (which works with the I2C printer motherboard)

klipper configuration:

```
[display oled_1]
display_group: _toolhead_0
lcd_type: ssd1306
i2c_mcu: EBBCan_0   # mcu EBBCan_0
i2c_bus: i2c1             # u'BUS_PINS_i2c1': u'PB6,PB7'
i2c_address: 60         # 0x3C
contrast: 230            # range from 0 to 256
```

Unfortunately, I still get a timeout error on the i2C.
I tried SDA/SCL both PB3,PB4 and PB6,PB7.

![ebb](https://u
[image]
ser-images.githubusercontent.com/64655979/184295294-7fe01e2e-0372-4e18-9494-0a29e100f540.PNG)
[image]

DrumClock, 2022-08-25:
pull up resistors are not connected to the i2C bus!
Even if I add them, there is still no communication between Kipper and EBB

**MCU 'EBBCan0' shutdown: i2c timeout**

[klippy (10).log](https://github.com/bigtreetech/EBB/files/9422083/klippy.10.log)

ldr, 2023-02-15:
hi, did you get this to work ? i am unable to use pb3/pb4 pins either on ebb42 board

DrumClock, 2023-02-17:
Hi @ldr 
it works fine for me on two EBB42 boards.
Do you have the current version of FW Klipper?

floodo1, 2023-06-19:
Also works fine for me (with recent firmware) when I specify the i2c pins:

```
[display ebb1106]
lcd_type: sh1106
i2c_mcu: EBBCan # The name of the micro-controller that the chip is connected to.
i2c_address: 60 # 0x3C
i2c_software_scl_pin: PB3
i2c_software_sda_pin: PB4
```

I believe `i2c_bus` cannot be used because pins PB3 and PB4 are defaulted to the SPI bus. Visible in the constants screenshot (under "BUS_PINS_spi3") and in Klipper source [I2C](https://github.com/Klipper3d/klipper/blob/master/src/stm32/i2c.c) and [SPI](https://github.com/Klipper3d/klipper/blob/master/src/stm32/spi.c)

Note: I ran into errors until I realized the i2c address on the silkscreen of my OLED screen was wrong!