bigtreetech/EBB #14: USB instructions

Wakinyantanka, 2022-08-28:
not seeing any instructions on connecting via USB. Specifically the Manta M8P
Any help would be appreciated.

Wakinyantanka, 2022-08-29:
Okay, I was able to figure it out so no response necessary.

closed 2022-08-29 by Wakinyantanka

honeybadgerotter, 2022-12-15:
Mind following this up with how you pulled it off? Heh. I’m in the same boat.

Wakinyantanka, 2022-12-15:
After opening your terminal and using the make menuconfig enable USB communication select PA11/PA12. Exit and hit yes when asked to save.
Then use the "make" command in terminal.

* [*] Enable extra low-level configuration options
* Micro-controller Architecture (STMicroelectronics STM32)
* Processor model (STM32G0B1) --->
* Bootloader offset (8KiB bootloader) * Clock Reference (8 MHz crystal)
* Communication interface (USB (on PA11/PA12)) --->
(Top)
Klipper Firmware Configure 
Enable extra low-level configuration options
Micro-controller Architecture (STMicroelectronics STM32)
Processor model (STM32G0B1) ---> Bootloader offset (8KiB bootloader)
Clock Reference (8 MHz crystal) --->
Communication interface (USB (on PA11/PA12)) ---> USB ids --->
(GPIO pins to set at micro-controller startup
[Space/Enter] Toggle/enter [Q] Quit (prompts for save)
[?] Help
[ESC] Leave menu
[/] Search
Press q to exit, and Yes when asked to save the configuration
All this is found in the M8P documentation on the Bigtreetech Hope this helps

Wakinyantanka, 2022-12-15:
*Bigtreetech GitHub