bigtreetech/EBB #15: What is the pitch of the Pin Header Jumper Cap, so I can order a replacement jumper cap

GadgetAngel, 2022-08-31:
What is the pitch of the Pin Header Jumper Cap for the EBB36 and EBB42 boards? 
My box spilled and I lost all the jumper caps. I need to order a replacement jumper caps but the jumper caps for the EBB36 are not 2.54mm pitch.  What is the pitch for these jumper caps?

gielbloks, 2022-09-01:
measure with a caliper / google pcb jumper spacings or something should be easy to find..

if you're in a rush.. carefully bridge them with a small drop of solder (one way action in most cases as it can be hard to remove an don't melt the plastic)

GadgetAngel, 2022-09-02:
> measure with a caliper / google pcb jumper spacings or something should be easy to find..
> 
> if you're in a rush.. carefully bridge them with a small drop of solder (one way action in most cases as it can be hard to remove an don't melt the plastic)

I tried the jumper cap with pitch 1.27mm but those are way too small.  I just order some with a pitch of 2mm and I will post if those worked when they arrive

gielbloks, 2022-09-02:
or just bridge it with a drop of solder..

ey wait . there are enough crimp contacts in the set.. crimp 2 together and use it as a bridge.. surely for the time being/solder the 2 crimp connectors together and you will have a jumper..

groeten/regards

Giel Bloks

Vergert 81
6662DX ELST