bigtreetech/EBB #16: What is max operating temperature?

bagobor, 2022-09-12:


rickyspanish8, 2023-01-28:
that would be pretty nice to know.

ValSimDim, 2023-04-09:
How can we get an answer to this? I can't find this info anywhere. Maximum operating temperature is important parameter of any electronic device. I'm also considering building printer with heated chamber.

BomberX, 2023-06-16:
Included STM MCUs on EBB boards cannot go higher than 85°C. BTT tries to mask the chip references, but they are STM32G0B1CBT6, while you have the STM32G0B1CBT3 that can go up to 125°C... But they have chosen the cheapest one, I suppose...

krzysiekslimak, 2024-09-22:
[image]
Have no idea what’s going on, should turn off for safety reason, and it doesn’t…

bigtreetech, 2024-09-22:
> ![IMG_0514](https://private-user-images.githubusercontent.com/87383241/369708542-926e46e4-b72f-4057-a1af-0c24a7cab951.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwMjkxOTcsIm5iZiI6MTcyNzAyODg5NywicGF0aCI6Ii84NzM4MzI0MS8zNjk3MDg1NDItOTI2ZTQ2ZTQtYjcyZi00MDU3LWExYWYtMGMyNGE3Y2FiOTUxLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyMlQxODE0NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzY3ZjM2NTBhNWE4ZTU2OTIyODNiYWNkZDExN2M3MDM5N2NiZmRlNDU2NDlkZjBmNTBmOGM0N2U4MTUzOTlhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.emeocvm2aGr6YciGB37_98dOYxnalrJjXPQLtqDCF1Y) Have no idea what’s going on, should turn off for safety reason, and it doesn’t…

The NTC is located beneath the FET and is designed to be a reading linked to that temperature. It is better to add the internal MCU thermistor so that you can read that directly. That should not exceed 85.

krzysiekslimak, 2024-09-22:
How can I read MCU temp from stm?

igeighty, 2025-01-06:
summer time here and with 55' chamber i see sb2209 into the 90's which sucks. would love BTT to clarify operating temperatures.

@krzysiekslimak 
i add these lines to my printer.cfg to read manta. cb2 and toolhead.
`[temperature_sensor Manta_8p]
sensor_type: temperature_mcu

[temperature_sensor CB2]
sensor_type: temperature_host

[temperature_sensor EBB_SB0000]
sensor_type: temperature_mcu
sensor_mcu: EBBCan`