bigtreetech/EBB #17: EBB42 CAN V1.2 flashing steps

Recoil1980, 2022-09-25:
Can someone please provide the exact steps for flashing the EBB42 CAN V1.2 board?  There is a mix of information for all the board versions, and it is not clear exactly what is applicable to this board.

Voodoobrew101, 2022-10-09:
Conect to pc with usb-c to usb-a cable. Open stm32 cube programmer.  Hold Reset on  ebb and tap boot (buttons may be backwards). Select usb in stm 32..there is a video on YouTube (by eddytheengineer?) now. He installs it with canboot.

Recoil1980, 2022-10-09:
I was finally able to flash it a week ago, but I had to use 3 of the videos he did on the other versions of the board, plus a couple of other guides, in order to compile a step-by-step list on adding this to my printer.  My plans are to hopefully push the steps once I am fully completed with my upgrade.

Headrush69, 2022-12-26:
> I was finally able to flash it a week ago, but I had to use 3 of the videos he did on the other versions of the board, plus a couple of other guides, in order to compile a step-by-step list on [...]

Any progress on this? Like you said, lots of differing information out there and obviously things have improved or changed. It seems like the EBB36/42 V1.2's only change is a pinout for the Hotend.

Voodoobrew101, 2022-12-27:
What do you mean. All the steps are in the pdf manual. Do you have a specific problem?