bigtreetech/EBB #19: EBB42 v1.0 EBBCan Timer too close error

danamaniac, 2022-10-06:
keep getting this shit error. Tried 250k, 500k and 1mbit can speeds. Using the octopus 1.1 in can bridge mode. tried different sd cards, power suplies. Unplugged camera. done updates and firmware updates. PLEASE HELP[
[image]
](url)
[Uploading klippy (5).log…]()

lanefloch, 2022-11-13:
Same here. Following.

ztzkao, 2023-03-26:
Same too.

369ELECRRIC, 2026-10-10:
> keep getting this shit error. Tried 250k, 500k and 1mbit can speeds. Using the octopus 1.1 in can bridge mode. tried different sd cards, power suplies. Unplugged camera. done updates and firmware updates. PLEASE HELP[ [image] ](url) Uploading klippy (5).log…

369ELECRRIC, 2026-10-10:
Gidday Dramamaniac,

Mate was so impressed by what you did man would love to chat with as I build for a hobby as would love to show you what I have done. But more so ask you a few questions can you text me on 0401 494 552. Based in Aussie. Ben

369ELECRRIC, 2026-10-10:
And Danamaniac my email is ben@financedepot.com.au 
Cheers 
Ben