bigtreetech/EBB #20: Connecting to BTT Hermitcrab CAN-Distribution Board via USB-C Destroys EBB42 v1.2 board

Voodoobrew101, 2022-10-06:
This happened to me 2/2 times. 

First attempt. I flashed CanBus Bootloader as with Communication interface (CAN bus (on PB0/PB1)) not paying attention thinking it was can-USB. Plugged in EBB42 to BTT Hermit Crab CAN-Distribution Board with USB-C cable and no jumper on VBUS, no other cables connected. Turned on printer and could not find EBB42 with "python3 flash_can.py -q" in putty.  From this point on Board lights up but will not enter DFU and connect to STM32CubeProgremmer. Multiple attempts to connect with canbus have also failed.

Second attempt. I immediately connected new EBB42 to BTT Hermit Crab CAN-Distribution Board with USB-C cable and no jumper on VBUS, no other cables connected. I did not flash board with any firmware. Turned on printer and heard a tiny pop before BL-touch from Hermit Crab extruder began to cycle. Two USB boards connected to outputs on BTT Hermit Crab CAN-Distribution Board. Port 1 had complete working BTT Hermit Crab extruder/hotend/bltouch/fans, Port 2 had EBB42 v1.2 board with no jumpers. After pop it appears that Component D6 is burned as well as some traces that connect to C31 look wet (probably heat damage?).

I understand that the setup directions have not been followed but why did this happen? It should not be so easy to destroy these boards. 

Is this because the wrong firmware is installed?
Is this because I did not connect VBUS jumper?
Is this because of the hardware from Hermit Crab is incompatible? 

My Setup.
Klipper on Raspberry Pi connected to and powered by J26 on BTT Octopus 1.1
BTT Octopus 1.1 runs X,Y,Z0,Z1 and bed heater
Raspberry Pi has BTT RPT-CAN HAT from Hermit Crab CAN
Hermit Crab BTT RPT-CAN HAT connects to Hermit Crab CAN-Distribution Board
Hermit Crab CAN-Distribution Board connects to power supply and Hermit Crab CAN-extruder with extruder/hotend/bltouch/fans connected

Voodoobrew101, 2022-10-09:
Ok, so you can't run 24v can to the usb-c. It will totally burn up the EBB.

I cut a 4 wire usb cable and wired it to the included can conector to make an adapter, it works great. Try to find a fat one with thicker wires.

I used a ohm meter to find which usb wire was which while probing the splitter board and referring to the 3d pdf schematic with traces.

Voodoobrew101, 2022-10-09:
The fact that you can't cross these usb-c cables from one of your devices to another should be in the manual with big red letters!

Benkult, 2023-03-24:
What's written on every hardware manual, is pinout.