bigtreetech/EBB #22: cannot connect bb42 v.1 to stm32 cube

antgore06, 2022-10-24:
trying to flash the ebb but all I get is no dfu detected. when plunging in to usb and holding the boot button and taping the rst button windows 11 sees dfu in fs mode but cube programmer will not find

wulin5500, 2022-11-29:
https://zadig.akeo.ie/#
[image]

AmigaZoid, 2023-04-19:
I had the same issue. I was using a 10 ft cable and tried every port on my computer but had the same issue.  I ended up getting it to connect using a older surface 2 pro that had the usb A connection with a different cable.  Not sure if this was the issue or virus protection is some way?  Good Luck