bigtreetech/EBB #24: Section 'tmc2240 extruder' is not a valid config section

victorlu0071, 2023-01-17:
the new SB EBB can board with 2240 TMC driver has some issues I get Section 'tmc2240 extruder' is not a valid config section on start

emersonsc, 2023-01-23:
As far as I know, theres no 2240 driver support in Klipper yet

fiferboy, 2023-01-27:
It looks like BigTreeTech has a branch of Klipper with TMC2240 support here: https://github.com/bigtreetech/klipper/tree/tmc2240

I haven't tested it yet and there is no merge request into Klipper yet, so it is possible that it isn't ready for use. I'll probably test it on my EBB SB2240 this weekend if I have a change.

victorlu0071, 2023-02-03:
just copy the file 2240 into the main branch and it worked

grimlokason, 2023-02-19:
It's kind of sad that there is no information about it in the manual...

pHetHH, 2023-02-27:
How to copy file 2240 into main branch or better use it?