bigtreetech/EBB #25: Build guide steel support wire details missing

zedn4c3r, 2023-01-23:
The build guide shows a steel wire is used to support the umbilical cable, and gives great details on how to bend it, but fails to say what size or type of steel wire to use.

dazkit, 2023-01-29:
I modified a remixed mod from umbilical that uses a 0.8mm Steel wire. You can download it here: https://www.printables.com/model/383128-voron-24r2-pg7-cable-gland-with-spring-steel-wire-

[image]

jcfain, 2023-10-29:
> The build guide shows a steel wire is used to support the umbilical cable, and gives great details on how to bend it, but fails to say what size or type of steel wire to use.

Yes im trying to figure out what gauge I should buy right now...

dazkit, 2023-10-29:
Piano wire with a diameter of 0.81 is recommended from many discord users