bigtreetech/EBB #28: TMC2240 undervoltage error

fabio011201, 2023-02-14:
with SB2240 after 20/30 minutes of printing I get the error 'tmc "extruder" reports error : GSTAT: 00000004 uv_cp=1 (undervoltage!)

both motor and board are new

victorlu0071, 2023-02-14:
i got the same issue on SB2240 but mine throw up error after just the first layer
it is a repeatable occurrence and happens when i switch to some hard filaments( larger extruder load)

fabio011201, 2023-02-14:
@victorlu0071
now that I know I am not the only one I am calmer.
My wiring does not rub against each other and is well secured.
Before I immediately blame the bigtreetech klippy folder, I am trying to find a problem in the configuration by doing several tests by changing the 'run_current' parameter.

(LDO motor nema 14 pancake, recommended run_current: 0.5. All tests are done with ABS in closed chamber, extrusion temperature 245°)

- tried setting run_current: 0.1, very little filament comes out of the extruder, almost imperceptible 5 minutes of printing no error (maybe test too short?)

- tried setting run_current: 0.75, correct amount of filament comes out, after 12 minutes of printing it reports error
"TMC 'extruder' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!)"
In the printout there does not seem to be any extrusion error, the extrusion is smooth.

- tried setting run_current: 0.4, the correct amount of filament comes out, after 1 minute of printing (2 layers) it reports error
"TMC 'extruder' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!)"
In the printout there seems to be no extrusion error, the extrusion is smooth.

- tried setting run_current: 0.9 the correct amount of filament comes out, after less than a minute of printing, layer 2, it reports again
"TMC 'extruder' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!)"
As before in the printout there seems to be no extrusion error, the extrusion is smooth.

- now I have tried setting run_current: 0.5 the correct amount of filament comes out, after less than a minute of printing, exactly where it stopped in the previous step, it still reports
"TMC 'extruder' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!)"
As before in the printout there seems to be no extrusion error, the extrusion is smooth.

- my Can cable is bundled together with the tube that brings the filament to the printer, assuming that the CanBus cable is bundled too tightly or that it is forced to move in strange positions I have tried removing all the clamps from the cable and leaving it longer, I have reprinted the same test while keeping run_current: 0.5 but the result does not change, I always get the same error after a few minutes of printing.
"TMC 'extruder' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!)"

- I now try to reprint the same part while keeping run_current: 05 and the cable without cable ties, but this time with the print chamber and the board cover open so as to rule out overheating problems, this time the print reaches the third layer but after a few minutes of printing again the error "TMC 'extruder' reports error: GSTAT: 00000004 uv_cp=1(Undervoltage!)" is reported.

All the tests performed are carried out under careful observation to check that the connections are not touching and rubbing against each other, considering that it is a CanBus module all the cables, except for the module, are firm and, considering that just restarting the printer makes everything work again, I exclude that there is any sheared, soldered or badly crimped cable

Since the result of my tests indicates that the problem persists initially after a few minutes of printing and then after a few minutes I cannot exclude a possible overheating problem, it would all have to be validated by re-running a test under the same conditions, with the printer cold, trying to print a material such as PLA so as to create much lower temperatures in the printing chamber (the board indicates that during printing with the chamber closed it reaches almost 60°, while the open chamber test indicates around 47°)

I'll stop here for today, as my next test I would like to try printing by connecting directly to the motherboard by reverting to powering the motor with the old TMC2209 to exclude the motor from possible problems

victorlu0071, 2023-02-15:
I did similar experiments
it all comes to the force that the nema14 motor pulls if i left the extruder gear open(latch not closed) which it pulls nothing (aka no force on motor) it can go indefinitely without triggering the under voltage. As soon as i put it back, no matter the current settings, it reports under voltage on 1-2 layers.
I switched back to some softer filament (PLA is actually hard) it can go further 2-3 layers before throwing error.
To be honest, i printed using the 2240 can board for almost a month only 1-2 occurrences, thought it is an overheating issue so i let the shroud on the stealthburner open which gives me another week of trouble free printing. But yesterday it happened again and the issue is persistent no matter what i did which renders the board unusable.

fabio011201, 2023-02-15:
@victorlu0071 ok then it will be a klippel problem, i guess there is nothing else to do but wait for an official update.
Thank you

victorlu0071, 2023-02-15:
i just sent the board back to BTT. I`ll update it here see what happened

fabio011201, 2023-02-15:
@victorlu0071 ok Thank you

msesterhenn, 2023-02-21:
Hey there,

I'm also affected.

A current of 0.8 works for me, currently testing 0.75. 0.7 would be safe for my extruder motor, however that still gives me the undervoltage error.

looxonline, 2023-02-21:
> @looxonline I am using E3D Revo Voron

Could you try to enter this into your extruder config section and see if it makes a difference?

driver_tpfd: 0

fabio011201, 2023-02-21:
@looxonline I am not home now, the next week I will try

looxonline, 2023-02-22:
@msesterhenn @victorlu0071 Please could you also try to enter this into your extruder config section. It should resolve the issue: driver_tpfd: 0

msesterhenn, 2023-02-22:
Hey there,

I have set the configuration to a current of 0.5 (the lowest ever for me) and your mentioned driver_tpfd: 0.
With these settings, I completed a 9 hour print tonight.

So far, looks good!

Maybe you could explain a bit more about this setting?

looxonline, 2023-02-22:
> Hey there,
> 
> I have set the configuration to a current of 0.5 (the lowest ever for me) and your mentioned driver_tpfd: 0. With these settings, I completed a 9 hour print tonight.
> 
> So far, looks good!
> 
> Maybe you could explain a bit more about this setting?

Good to hear! Here is a bit more info:

https://discord.com/channels/460117602945990666/825356024130175026/1077859127012888586

victorlu0071, 2023-02-23:
FYI my board is at BTT now im unable to test it for now. My extruder is a genuine dragon HF extruder and my filament path is quite long The error happens after i upgrade my extruder gears to hardened steel and improved my filament spool holder design. it all comes down to the extruder load i believe.

victorlu0071, 2023-02-23:
its a good and in depth insight on the issue on voron design channel. i`ll test it as soon as i get the board back from BTT

looxonline, 2023-02-23:
> FYI my board is at BTT now im unable to test it for now. My extruder is a genuine dragon HF extruder and my filament path is quite long The error happens after i upgrade my extruder gears to [...]

Ah OK. It is your unit that we have for testing. We are working with it currently. BTW, I wrote the post on the voron channel.

victorlu0071, 2023-02-23:
darn good insights on the topic. Glad i can be part of the debugging task force. Regarding the can cable issue, im not sure if my cable is all good because i will occasionally get the missed scheduling of the next digital out event issue using the 2240. Only happens on first layer though. if it did not throw the error on layer 1 the print can be done successfully. When it happens, it requires a full shutdown cycle and wait at least 1 min for the capacitors to drain then i can print normally. I believe it has something to do with either static charge on the PEI sheet or something on the EMF side when nozzle touches the bed which causes the issue, or im just getting a faulty cable?

looxonline, 2023-02-23:
> darn good insights on the topic. Glad i can be part of the debugging task force. Regarding the can cable issue, im not sure if my cable is all good because i will occasionally get the missed [...]

That is really interesting. It could be due to static buildup coming through the thermistor or heater wires and into the board. I'll check out the sch to see what path it may be taking.

eddiedellz, 2023-03-04:
Same issue here. Ive tried all the above fixes and the tmc2240 klipper branch still same after 1st layer

fabio011201, 2023-03-09:
@looxonline  @eddiedellz sorry if I answer now but in the meantime I had broken the Revo voron E3D and I replaced it with the rapid HF also with "driver_tpfd: 0" I have the same error.

nymica, 2023-03-30:
I've had this same problem consistently when printing with PLA, however I've now loaded up a roll of ASA and haven't had a single failure and it prints beautifully. It looks like my ender will be reserved for PLA ol.

looxonline, 2023-03-30:
> I've had this same problem consistently when printing with PLA, however I've now loaded up a roll of ASA and haven't had a single failure and it prints beautifully. It looks like my ender will be reserved for PLA ol.

Did you try the tpfd tweak in the config section?

nymica, 2023-03-30:
> > I've had this same problem consistently when printing with PLA, however I've now loaded up a roll of ASA and haven't had a single failure and it prints beautifully. It looks like my ender will be reserved for PLA ol.
> 
> Did you try the tpfd tweak in the config section?

Yep but it still persisted with PLA.

Taaaserfaaace, 2023-04-10:
I just experienced this issue last night. I completed the initial installation of the SB2240, run a few test prints, due to installing a new Hotend(Voron Dragon HF) and after a few hours of continuous printing I received this error. Not sure why. I included the updated tpfd in the config

Rhodani, 2023-04-13:
Ok, this is what I have done. 

I have fixed the power wiring so I am 100% sure that this is not a problem with PSU or anything else. 

The printer has worked perfectly ever since I built it. 
The Undervoltage error for the extruder has to be a problem caused by the SB2240 installation. 

I have put the driver_tpfd:0 in my config file. I have checked resistance on the extruder motor. Stable at 2.78 ohm on both pairs, even with heavy wiggeling of the wires.

If I raise the extruder current to 0.65 the print goes on for longer but fails after 10-20min. With a current of 0.5 it fails earlier aprox. 5-10 min. 

I have used almost two full days getting CAN BUS up and running and this is the only problem that seems unsolvable. I run the BTT SB2240 and only after I installed this board I have this problem.

I have installed a fan directly at the PI 3B to lower temps and a 2020 fan on the toolhead to keep the SB2240 board cooler. Without it gets close to 90 degrees Celsius but with the fan it did not go over 70.

I have checked Voltage on the PSU and on the toolhead. PSU and toolhead reads 24.2V with low load. With full load on bed and hot end it drops to 23.7V.

What do I do now? I have a printer that I can not use and it will take me a VERY long time to go back to regular cables again. Not to mention the cost of the parts to make this upgrade. 

Please BTT, give me a straight answer here. I am not the only one having this problem. Do you have a solution or do you actually have no idea what is causing this?

My printer is a LDO 350 with rapido UHF. All other parts are standard for that kit and has been working VERY well since day one.

looxonline, 2023-04-13:
> Ok, this is what I have done.
>
> I have fixed the power wiring so I am 100% sure that this is not a problem with PSU or anything else.
>
> The printer has worked perfectly ever since I built it. The [...]

There are other users who are experiencing this but they are very few in number. We have been actively investigating the issue with each one of them to try and figure out whether there is a common thread. In one case, the user had unknowingly purchased fake meanwell PSUs and a bad SSR. There was 3Vp-p AC sitting on top of his DC rails that he was only able to measure with a scope. Replacing the PSUs and the SSR fixed the issue. 

In another case, a user removed the board, shortened the wiring, reinstalled the board, and all was well.

It seems as if the TMC2240 is extremely sensitive to ripple on the supply rail and so it is likely that there is some sort of ripple on your rail. The fact that your rail drops so much with a full load on the bed and hotend seems to support this. You should not be seeing a 0.5V drop at your PSU. That is an indication that your PSU is unable to maintain regulation under full load.

If you use discord I would suggest that you send a message to @BTT University on the bigtreetech channel on the voron server so that you can discuss this further.

Rhodani, 2023-04-13:
I have sent several messages on Discord but have not heard back? This is
driving me mad. I have had Zero problems untill I installed this board.

The PSU is from an LDO kit so I highly doubt that its a fake. And should I
not have had problems before since I used the same PSU then?

Best regards
Jon

looxonline, 2023-04-13:
> I have sent several messages on Discord but have not heard back? This is driving me mad. I have had Zero problems untill I installed this board. The PSU is from an LDO kit so I highly doubt that its [...]

Were you wiring directly to the hotend without a CAN board before? If so, then the SB2240 is out of the picture and that explains why you never had issues. As I mentioned, that particular IC from TMC appears to be very, very sensitive to spikes or dips on the rail. We will be happy to swap it out but we have tried that with other users and it did not help until they figured out what the issue was on their rail.

Rhodani, 2023-04-13:
> > I have sent several messages on Discord but have not heard back? This is driving me mad. I have had Zero problems untill I installed this board. The PSU is from an LDO kit so I highly doubt that its a fake. And should I not have had problems before since I used the same PSU then? Best regards Jon tor. 13. apr. 2023, 11:30 skrev Luke Harrison _**@**_._**>:
> > […](#)
> 
> Were you wiring directly to the hotend without a CAN board before? If so, then the SB2240 is out of the picture and that explains why you never had issues. As I mentioned, that particular IC from TMC appears to be very, very sensitive to spikes or dips on the rail. We will be happy to swap it out but we have tried that with other users and it did not help until they figured out what the issue was on their rail.

Ok, so if I understand correctly, you are saying that you think my problem is related to the Meanwell PSU? And that I should not have a drop when I draw 160ish Watts with the hotend. this might make the TMC2240 go into undervoltage protection.

So I have to try to buy a new PSU to test if thats the problem? Or is there another solution that I can try that wont cost me more money and time? 

Is there some way to test the ripple or the unstable power coming out of the PSU?

Rhodani, 2023-04-13:
Directly to the PSU.

Rhodani, 2023-04-16:
> > I have sent several messages on Discord but have not heard back? This is driving me mad. I have had Zero problems untill I installed this board. The PSU is from an LDO kit so I highly doubt that its a fake. And should I not have had problems before since I used the same PSU then? Best regards Jon tor. 13. apr. 2023, 11:30 skrev Luke Harrison _**@**_._**>:
> > […](#)
> 
> Were you wiring directly to the hotend without a CAN board before? If so, then the SB2240 is out of the picture and that explains why you never had issues. As I mentioned, that particular IC from TMC appears to be very, very sensitive to spikes or dips on the rail. We will be happy to swap it out but we have tried that with other users and it did not help until they figured out what the issue was on their rail.

Ok so I have some more info now. I have printed a couple of days after I swapped out the extruder motor with a new identical motor. It has been printing at speeds up to 150 mm/s, accel up to 5000, with an 0.2 layer hights 0.4 with 0.4 nozzle bronze. Since I have upgraded other parts on my machine I wanted to go faster. I tried a run with 350mm/s and 6000 accel. I use a Rapido UHF hotend. Now I get the error again. So I am back to where I startet pretty much. If the chip is this sensitive, you need to come up with some sort of fix. My setup is a pretty regular setup and having a 2240 that is so prone to throw undervoltage errors left and right is just not a good way to go. 

Will I fix the problem if I go for a SB2209? Since that driver is less sensitive?

Right now the whole CAN BUS experience just sucks and that is sad.

Taaaserfaaace, 2023-04-17:
> > > I have sent several messages on Discord but have not heard back? This is driving me mad. I have had Zero problems untill I installed this board. The PSU is from an LDO kit so I highly doubt that its a fake. And should I not have had problems before since I used the same PSU then? Best regards Jon tor. 13. apr. 2023, 11:30 skrev Luke Harrison _**@**_._**>:
> 
> > > […](#)
> 
> 
> > 
> 
> > Were you wiring directly to the hotend without a CAN board before? If so, then the SB2240 is out of the picture and that explains why you never had issues. As I mentioned, that particular IC from TMC appears to be very, very sensitive to spikes or dips on the rail. We will be happy to swap it out but we have tried that with other users and it did not help until they figured out what the issue was on their rail.
> 
> 
> 
> Ok so I have some more info now. I have printed a couple of days after I swapped out the extruder motor with a new identical motor. It has been printing at speeds up to 150 mm/s, accel up to 5000, with an 0.2 layer hights 0.4 with 0.4 nozzle bronze. Since I have upgraded other parts on my machine I wanted to go faster. I tried a run with 350mm/s and 6000 accel. I use a Rapido UHF hotend. Now I get the error again. So I am back to where I startet pretty much. If the chip is this sensitive, you need to come up with some sort of fix. My setup is a pretty regular setup and having a 2240 that is so prone to throw undervoltage errors left and right is just not a good way to go. 
> 
> 
> 
> Will I fix the problem if I go for a SB2209? Since that driver is less sensitive?
> 
> 
> 
> Right now the whole CAN BUS experience just sucks and that is sad.

I've had the same issue with the SB2240. After seeing there was no specific fix to this issues I decided to swap the SB2240 with the SB2209. Since switching I have not had an issue since. Being the TMC2209 runs a little hotter, I installed a 3010 fan on the SB to keep it cool. The TMC2209 chip now hovers around 65c to 70c with a chamber temperature of 55c to 60c while printing ABS. I also lowered my run current to 0.450 with a hold current of 0.300. Hope this helps.

Rhodani, 2023-04-18:
I also took new measurements on the PSU and I now measure zero drop when the hotend is at 100% so that kind of rules out the PSU. I really dont know what to test now. I am talking to another individual on the discord server and we have tried a lot of stuff. And nothing works. So I am probably left with returning it and hopefully the 2209 will work better :(

OldGuyMeltsPlastic, 2023-04-18:
I am also running into this same extruder undervoltage error today with a BTT SB2240, with a Rapido UHF hotend. Mine is a Formbot V2.4R2 kit with a genuine Meanwell power supply.
This same printer was previously working for months with a Mellow Fly SB2040, but that board was very fragile, so I decided to upgrade to the BTT SB2240, figuring it would be more robust.
I also have a spare EBB36 that I could try instead, but that would mean re-crimping all of the toolhead wiring again, so I am trying to avoid that.
I have just now added `driver_tpfd: 0` to my extruder config, and will test if that makes any difference.

OldGuyMeltsPlastic, 2023-04-19:
No change in behavior with `driver_tpfd: 0` applied to the config.
I have `run_current: 0.6` set.

OldGuyMeltsPlastic, 2023-04-19:
Am wondering if a different hotend would make a difference. The Rapido is known for drawing more power initially when heating up, but these errors happen minutes into the print, so that seems unlikely to be a contributing factor.

Surion79, 2023-04-20:
I have the same issue with the tmc2240. It happens sometimes and there is no indication why. i print the same stl multiple times and sometimes its fine, other time it just stops right within the print. not first layer or so
It is frustating to print larger models just to put it in the waste because of this inconsistency

Rhodani, 2023-04-20:
Ok so it seems that there are actually a good number of users having this
problem 👍 Thx

deadlydayne, 2023-04-22:
Recently installed 2240 board here as well (3 weeks ago).  Printed for a week and a half no problems, now all of a sudden tonight, I cant get past first or second layers without this error.  Nothing is different from these 3 weeks.  Just literally decided to start erroring out.
My first canbus setup and its not left me happy

jtroutt19, 2023-04-22:
I am also having this error all the sudden. probably the 9th print since installing the sb2240. failed after almost an hour of solid printing for the first time. Frustrating wasting that much filament for sure. driver_tpfd: 0 is already include in my extruder config and my PSU is holding voltage just fine. 

BTT needs to fix this!

deadlydayne, 2023-04-22:
I've tried all suggestions in this thread.  Even chopped about 2.5 feet of my umbilical, and nothing changes.  
I can pre heat the machine, change temps, move the head around and everything is fine.  But as soon as it starts to extrude anything I can watch the temps of the driver spike up from 40c right to 60c and then shortly after it will error out.
I've been curious if going to the 2209 would fix this, I'm about to spend the money on another board because of it, but I dont want to have to waste money on 2 products if they cant get this one right.  This definitely doesn't seem to be a small user case issue.  

This thread has been going since early Feb.  With more and more people jumping in every day making the same complaints.  Idk what BTT be doin, but it doesn't feel like we are really getting support for the product we purchased here.

jtroutt19, 2023-04-22:
Bit of an update. After lowering my print speeds I was able to complete a print that had been failing. Failed a total of 5 times. I believe the issue happens when the extruder has a higher load on it. Heat may also play a role here. I ran about 5 degrees cooler on the sb2240 and the stepper.

OldGuyMeltsPlastic, 2023-04-22:
Define "lower speeds"? What nozzle and bed temps are you using, and what
kind/brand of filament?

jtroutt19, 2023-04-23:
> Define "lower speeds"? What nozzle and bed temps are you using, and what kind/brand of filament?
> […](#)

I am using a .4mm nozzle. Currently printing Overture PETG. 85 bed temp, 240 nozzle temp. max print speed of 120mm/s

jtroutt19, 2023-04-23:
Found some info in the Datasheet.

The TMC2240 features a UVLO protection for VM, VCC_IO, and the charge pump. 
UVLO condition on VM is triggered below 4.15V (max). 
UVLO condition on VCC_IO is triggered below 1.95V (max). 
UVLO condition on the charge pump is triggered in case of an error condition of the charge pump, e.g., due to a wrong 
capacitor value. 
A VM UVLO condition can be read from register GSTAT as flag vm_uvlo. This flag is a write-clear flag. It must be actively 
set to 1 to clear it. The UVLO condition is also shown at the DIAG0 pin depending on the configured pin settings. 
During a VCC_IO UVLO, no communication with the IC is possible. DIAG0 pin will be active low (open drain). 

so a simple python script should tell us where this undervolt is happening.

LambdatheCat, 2023-04-26:
I also have this error, I have been printing fine for a while now using ABS with high tempuratures and speeds but when I switched to PLA+ with a temp of 220 and a speed of 200mm/s I keep getting this error within the first few layers, I have made sure my wirring is perfect and I included `driver_tpfd: 0` but it still made no difference, I am going to try the same print but slower to see if it makes a difference. Would love for this issue to be solved though as I dont like the idea of wasting lots of filament due to a stupid inconsistancy!

deadlydayne, 2023-04-26:
Putting this: SET_TMC_FIELD STEPPER=extruder FIELD=THIGH VALUE=1048575
in my Print_Start macro (inside your printer.cfg file) solved my issues.

LambdatheCat, 2023-04-26:
> 

once my current print completes (or fails) I will give this a go, but at this current moment the slower print seems to be working correctly. once its over I will try this out and see if I can go back to printing at 200mm/s.

Thanks for the info!

deadlydayne, 2023-04-26:
User Andrewmcgr in the discord listed his entire Extruder calibration.  Some have copy/pasted it as theirs and changed some settings such and pins and what have you to match whats needed for your machine.  
I havent checked back in 2 days since mine has been running good, but Id say keep an eye on Canbus Depot in the discord and keep an eye for Andrewmcgr.

LambdatheCat, 2023-04-26:
The print I had working ended up failing and I tried it again with this `SET_TMC_FIELD STEPPER=extruder FIELD=THIGH VALUE=1048575` in my start macro but the print still ended up failing. I dont know what to do, might have another look tomorrow. :/

gx1400, 2023-05-01:
https://github.com/Esoterical/voron_canbus/blob/main/toolhead_flashing/common_hardware/BigTreeTech%20SB2209%20and%20SB2240/SB2240%20Stepper%20Configuration%20and%20Undervoltage.md

Rubikgood, 2023-05-03:
I have the same problems with SB2240. The board worked fine for 2 days, but now there is an error on the first layer. I checked and even replaced the power supply, checked all the wiring and the quality of the crimp. Obviously, the product is raw and we act as beta testers.

baracoder, 2023-05-03:
I am also affected by this: It was working fine for two weeks printing with ABS, but switching to PLA yesterday, I got this error. 
A few notes:
* I did run a volumetric flow test-print with ABS and tested different `run_current` values until the extruder skipped steps without being affected by this error
* The extruder motor (Moons) stays cool when printing PLA
* I have tried lowering the print speed by 50% without success.
* My config already contains `driver_tpfd: 0`

The part cooling fan is not running on 100% with ABS prints, but it does with PLA. Could the fan speed have an effect on this issue?

jtroutt19, 2023-05-03:
@baracoder i would think it would but give it a try and see. Report back

LambdatheCat, 2023-05-04:
That's an interesting variable, all the fails from what I can see are caused by printing PLA as the board works fine with ABS, could it be an issue with fan speed? Or that PLA is harder or something? We will see when baracoder runs a test. I will try later today as well to see if it makes a difference

Rhodani, 2023-05-04:
No, There are several that has the same problem with ASA, ABS and other
materials 👍

LambdatheCat, 2023-05-04:
> […](#)

Never mind then lol. Seems to just be unreliable then. I only thought that because mine works with ABS but when I switch to PLA I get the error.

Rubikgood, 2023-05-05:
> https://github.com/Esoterical/voron_canbus/blob/main/toolhead_flashing/common_hardware/BigTreeTech%20SB2209%20and%20SB2240/SB2240%20Stepper%20Configuration%20and%20Undervoltage.md

I have tried these settings. It has become a little better, now 1.5 layers are printed before error instead of 0.5. But the remaining 254.5 layers remained unfinished :) And no reasonable comments from BTT, they continue to sell the raw product

Rubirosa007, 2023-05-08:
i want to throw this thing in the trash...

gabtremblay, 2023-05-08:
- Verified that my Meanwell psu is genuine
- Verified that my SSR's are genuine
- Re-did all my CAN wiring
- Applied all solutions found in Esoterical git.

Still getting the error, more and more frequently. 

My setup ran fine for months, then it suddenly started.

deadlydayne, 2023-05-08:
Mine has started receiving the error again as well. Ordered 2209 driver
going to replace. Can’t have this printer down this often or so finicky
about what material it “wants” to print.

[247 more replies]