bigtreetech/EBB #31: EBB SB2240_2209 CAN Module mount STE/STL is missing

irebane, 2023-02-27:
Build manual referring to use special little mount between SB2240 and CW2 extruder. 
Could you please add it to the repository or add details to the manual where to download it. 

Thanks.

dazkit, 2023-02-28:
The sb2209/sb2240 is an inspiration from hartk Stealthburner Toolhead Board. The spacer is from hartk
https://github.com/hartk1213/MISC/blob/6e4f22c85e541eafab4471c52c59f51c7ac6b2b8/PCBs/Stealthburner_Toolhead_PCB/STLs/pcb_spacer.stl

dazkit, 2023-02-28:
and of course this is missing in the manual

karamfil, 2023-04-21:
Still missing, took me 20 minutes to figure out...

ja2ui0, 2023-08-29:
Would you believe it's still missing more than six months later? I was completely confused as to why the SB main body would not line up with CW2. Please add this part to the repo!

travisdieckmann, 2023-10-24:
Still missing from the repo...

IsoBaba, 2024-07-14:
And it is still missing... Thank you @dazkit.