bigtreetech/EBB #32: EBB 42 V1.2 bricked!

GrNeOp, 2023-03-11:
Hi all,
Here is my scenario. Manta M8P with CB1 + U2C + EBB 42 V1.2 .
Manta +CB1 with Klipper installed. U2C configured, and I tried to follow the BTT [Video](https://www.youtube.com/watch?v=GaMOe1yirPk) to install the firmware on the EBB 42 V1.2. Once the board was in DFU mode and connected to STM32 Cube Programmer, I clicked Download and got an error 

`15:06:27 : STM32CubeProgrammer API v2.13.0 | Windows-64Bits 

  15:07:04 : Read File: \\Mac\Home\Desktop\klipper.bin

  15:07:04 : Number of segments: 1

  15:07:04 : segment[0]: address= 0x0, size= 0x67CC

  15:09:29 : UR connection mode is defined with the HWrst reset mode

  15:09:29 : USB speed   : Full Speed (12MBit/s)

  15:09:29 : Manuf. ID   : STMicroelectronics

  15:09:29 : Product ID  : DFU in FS Mode

  15:09:29 : SN          : 204D39545542

  15:09:29 : DFU protocol: 1.1

  15:09:29 : Board       : --

  15:09:29 : Device ID   : 0x0467

  15:09:29 : UPLOADING OPTION BYTES DATA ...

  15:09:29 :   Bank          : 0x00

  15:09:29 :   Address       : 0x1fff7800

  15:09:29 :   Size          : 68 Bytes

  15:09:29 :   Bank          : 0x01

  15:09:29 :   Address       : 0x1fff7870

  15:09:29 :   Size          : 4 Bytes

  15:09:29 : UPLOADING ...

  15:09:29 :   Size          : 1024 Bytes

  15:09:29 :   Address       : 0x8000000

  15:09:29 : Read progress:

  15:09:29 : Data read successfully

  15:09:29 : Time elapsed during the read operation is: 00:00:00.025

  15:09:31 : Memory Programming ...

  15:09:31 : Opening and parsing file: klipper.bin

  15:09:31 :   File          : klipper.bin

  15:09:31 :   Size          : 25.95 KB 

  15:09:31 :   Address       : 0x08000000 

  15:09:31 : Erasing memory corresponding to segment 0:

  15:09:31 : Not flash Memory : No erase done

  15:09:31 : File segment @0x08000000 is not 255-bytes aligned. It will be aligned to @0x07FFFFF8

  15:09:31 : Download in Progress:

  15:09:31 : Error: failed to download Segment[0]

  15:09:31 : Error: failed to download the File` 

Now the board has the green and blue light (fainted) and I can't connect to it anymore. The cube programmer or the Manta M8P with CB1 don't see it anymore.
I don't know what to do...
Any help would be appreciated.
Thank you!

CybeRNerO, 2023-03-11:
I was struggling for 5 hours today with the same issue.
I tried to set this up today and I got the exact same error prompt on SMT32CubeProgrammer.

FlorinAnghel, 2023-03-13:
I had the same problem... but this fixed everything:
https://maz0r.github.io/klipper_canbus/toolhead/ebb36-42_v1.1.html

GrNeOp, 2023-03-13:
> I had the same problem... but this fixed everything:
> https://maz0r.github.io/klipper_canbus/toolhead/ebb36-42_v1.1.html

Thank you! I just start a 3 days business trip. I will try once I’m back and will report back.

looxonline, 2023-03-20:
> Hi all, Here is my scenario. Manta M8P with CB1 + U2C + EBB 42 V1.2 . Manta +CB1 with Klipper installed. U2C configured, and I tried to follow the BTT [...]

This is not a hardware issue. For some reason, the STM programmer is not liking the files that are being generated by the klipper compiler. It is offsetting them so that they sit in an undefined memory segment. I have not had time to look into how to correct this but if you use dfu util to program then it does a good job as others have mentioned.

gurlen, 2023-04-07:
I had this same issue, and it appears to be an issue with the latest version of STM32CubeProgrammer (2.13.x) - I uninstalled 2.13 and installed 2.10.0 and it worked perfectly.

kjburto79, 2023-04-13:
> 
This worked for me as well!

selrahc13, 2024-03-04:
Latest STM32CubeProgrammer (2.15.0) seems to have fixed this issue.