bigtreetech/EBB #34: Manta E3EZ+CB1+EBB36 v1.2

alexandrugai, 2023-03-15:
hi there...i need some help. yesterday arrived the ebb36 v1.2 and i'm trying to connect it to manta e3ez over can. manta flashed with klipper and selected "usb to can bus bridge (usb on PA11/PA12 and canbus on PB8/PB9)" and here begins the problem... when flashed this way, CB1 doesn't recognize manta anymore over "usb serial" connection. can0 network appears, but just the manta, not the ebb36. to mention that when i try to flash the ebb36 with stmcube i get an error "download failed" like in issue #32  or something like this but when i flash it with canboot, flashes ok (at least no error on download). in the manta manual i didn't found nothing about can bus. after first attempt to flash the ebb, it remains with one led full lit and the other one barely visible.

FlorinAnghel, 2023-03-16:
do this tutorial... it works without failure
https://maz0r.github.io/klipper_canbus/toolhead/ebb36-42_v1.1.html

alexandrugai, 2023-03-17:
yes...it works now. i was flashing the manta with wrong can pins under canbus comms. thank you

closed 2023-03-17 by alexandrugai

nicolas-lang, 2023-07-15:
Would you mind sharing your final cfg?
I have a similiar issue and cant find the ebb.
- Which pins did you select for Canbus in the Manta E3EZ-klipper-menuconfig
- Which physical connector did you use on the Manta E3EZ? The left one, labeled tx pb13/rx pb12 or the right one labeled canH/canL
[image]

maxcalavera81, 2025-03-22:
Good.  I also have this doubt.  Which connector on the blanket did you connect the ebb36 to?  Thank you very much in advance.