bigtreetech/EBB #36: Unable to read tmc uart 'extruder' register IFCNT

dogtrpp, 2023-03-19:
Replaced an EBB36 with the SB2209 and getting this error when I try to extrude.

"Unable to read tmc uart 'extruder' register IFCNT"

Extruder config (I changed the MCU name from EBBCan to can0)

## EXTRUDER 
[extruder]
step_pin: can0: PD0
dir_pin: !can0: PD1
enable_pin: !can0: PD2
##  Update value below when you perform extruder calibration
##  If you ask for 100mm of filament, but in reality it is 98mm:
##  rotation_distance = <previous_rotation_distance> * <actual_extrude_distance> / 100
##  22.6789511 is a good starting point
rotation_distance: 22.68486
gear_ratio: 50:10 #for CW2
microsteps: 16
full_steps_per_rotation: 200 #1.8deg Motor
max_extrude_only_distance: 200.0
max_extrude_cross_section: 50.0
max_extrude_only_velocity: 75.0
max_extrude_only_accel: 1500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: can0:PB13
sensor_pin: can0:PA3
sensor_type: ATC Semitec 104NT-4-R025H42G
#pullup_resistor: 2200             # this is the default for the Revo heater
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 327
min_extrude_temp: 170
pressure_advance: .035

## EXTRUDER MOTOR
[tmc2209 extruder]
uart_pin: can0: PA15
stealthchop_threshold: 999999
run_current: 0.650

closed 2023-03-20 by dogtrpp

travisdieckmann, 2023-10-24:
I'm experiencing the same error and I see you closed the issue... What was the fix?

dogtrpp, 2023-10-24:
> I'm experiencing the same error and I see you closed the issue... What was the fix?

It's pretty silly. I thought I bought the version with the 2209 driver, but turns out it was the 2240. Changed the configuration in Klipper and it worked fine.

travisdieckmann, 2023-10-24:
🤦 I did THE SAME EXACT THING! 🤦

PkBrasil17, 2024-08-18:
and i did ... the exact same thing!!!!