bigtreetech/EBB #37: Connector of SB2204

irgusite, 2023-03-20:
Hi, I wanted to know what connector is used for thermistor/endstop on the ebb2209.

I see it is a 1.25mm pitch, but no info for finding a compatible crimping tool or more connectors...

Thanks for the info

BerzerkaDurk, 2023-03-22:
They're JST PH connectors.
[image]
[image]

irgusite, 2023-03-22:
JST PH is 2mm pitch. And searching for jst 1.25mm results in at least 10 different connectors.....

BerzerkaDurk, 2023-03-25:
So, I've done a bit of research. These 1.25 connectors are Molex PicoBlade clones. Someone confirmed this on Reddit, and when you open BTT's STEP file of the board in Fusion 360, the connectors are labeled MX1.25 connectors. Google also says that these are compatible with PicoBlade connectors from Molex.

irgusite, 2023-03-25:
Ah! Thanks a lot for the research! I will test with some picoblades from work to confirm :)

BerzerkaDurk, 2023-03-26:
I'm happy about it because I've been dreading trying to crimp these little buggers, and Digi-Key has tons of pre-crimped PicoBlade leads available.

miracuruzrh, 2023-08-25:
Would be happy to see the manual updated with the information that this are Molex Picoblade connectors. JST-GH is not correct. I guess somewhere has occured a mixup, because also some resellers sell JST-GH which are in reality Molex Picoblade. Maybe someone made a clone and just labeled them wrong, I guess.