bigtreetech/EBB #39: Question: What is the round screen shown for the SB2209/2240 Board?

astrof, 2023-03-22:
On the page for the SB2209/2240 board there's a picture of an LCD screen somehow attached to the CAN port on the board; what screen is this and how does one attach it?

truelecter, 2023-04-05:
Did you manage to find this thing?

astrof, 2023-04-25:
So my closest guess is it might be one of those [Waveshare Round Displays](https://www.amazon.com/Display-Arduino-Raspberry-240%C3%97240-Embedded/dp/B095X1JDCM/ref=sr_1_4?keywords=waveshare+round+display&sr=8-4), they have both with an RP2040 built in and one that is just SPI, but I'm guessing the SB cover has to be modified a bit to allow for the connector.  I haven't found one that would connect to the CAN port but the SPI one could I guess be wired to the SPI connectors.  I'll probably keep this as a side project for later (when hopefully there will be something more "official").  Disappointing that @bigtreetech hasn't mentioned anything specifically.

truelecter, 2023-04-25:
So I've checked a little bit more and it appears that LCD screen is a BTT leak, which was not supposed to be in the SB2209/2240. 
Latest updates on this: https://www.facebook.com/530918831/videos/194441013221532/

And yes, looks like there will be custom SB cover with special vents for the blower

truelecter, 2023-04-27:
And here it finally is!
https://biqu.equipment/products/bigtreetech-knomi-v1-0

astrof, 2023-04-29:
Oh neat! The sizing looks similar to the waveshare screens just with an ESP32 instead of an RP2040.  I wonder if the round LCDs with built in RP2040s would fit the custom piece.  Of course now I have to remix my SB cover that already was a remix to accommodate a nozzle cam lol.