bigtreetech/EBB #41: SB2209 Electrical Burning Smell

sergelhr, 2023-04-05:
After moving from tap mechanical switch to Voron OptoTap PCB v2.4.1 5-24V, an electrical burning smell was evident.
opto tap indication red/blue was ok. i didn't select 5 or 24v pad at the back of the pcb !
After returning to  mechanical switch no more smell (already burnt!), no evidence of heat damage onto opto tap or EBB, the only symptom rgb leds remains off! nil time so far to carry out a trouble shoot, might be front face was not in line with the connector ?

sergelhr, 2023-04-21:
after replacing the SB2209 pcb only, fault still present, leds remains off, after using a new EBB SB0000 CAN, and the new SB2209, LED working again.
Probably one chip linked to rgb port on SB0000 ,has burnt out. fixable ?

MyStoopidStuff, 2023-04-22:
I'm waiting on my EBB SB2209 to arrive, and also will be using the Optotap 2.4.1 board, so I am curious on how you set yours up.  It sounds like you are using the Optotap 2.4.1 using VCC (24v) and not 5v, since you mentioned you did not short the jumpers on the OptoTap, is that correct?  Were you connecting the OptoTap 2.4.1 to the proximity input on the SB2209 (and if that is the case, how was your proximity voltage jumper and npn/pnp jumpers set)?  Thanks!

sergelhr, 2023-04-22:
#MyStoopidStuff
voron tap wiring refer to 
https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/Build%20Guide/EBB%20SB2240%202209%20CAN%20v1.0%20Build%20Guide.pdf
page 17 (21 of 45)
And yes regarding optotap 2.4.1,for peace of mind, i cut the trace (24v) and make a bridge to 5v see info below
https://lab4450.com/product/voron-tap-probe/

[image]

sergelhr, 2023-04-22:
Just another issue with my new sb2209 and firmware klipper v0.11.0-180-g011b4e39 no more extrusion !
klippy log-> TMC extruder failed to init: Unable to read tmc uart 'extruder' register IFCNT
I've reverted to my previous sb2209 with firmware klipper Version: v0.11.0-148-g52f4e20c , no issue with extruder and led ok, so for sure the EBB SB0000 CAN was my issue.