bigtreetech/EBB #45: SB2209 heater cartridge outout only 1.5V ?

Danman234567, 2023-04-19:
My heater cartidge dosent want to heat after an update of system files on mainsail, when activated it only output 1.5V ? it should be 24V for the heater ? i dont know where to look , to a i need to reflash the toolhead to ''update'' it ?

nilchaos, 2024-03-15:
I am experiencing a similar issue but with a new build - I flashed katapult and then klipper according to the supplied instructions.

Despite everything else working just fine (RGB addressable, thermistor reading, CAN communication etc) the fans and thermistor don't appear to function.

Multimeter reading across HE0 (PB13 / GND) shows a reading of around 1V.

[image]

Quilsen, 2024-03-15:
I have the same issue, but on two boards that bizarre