bigtreetech/EBB #47: TMC 'extruder' reports error: GSTAT:     00000001 reset=1(Reset)

Aleishus, 2023-04-24:
I use EBBSB 2209 ，how to solve it

Taaaserfaaace, 2023-05-05:
I recently started having this issue today after several weeks of using my SB2209 with zero issues, I checked Discord for a possible solution and found that two other users had the same issue which started today. Not sure if this is due to a recent killer update? But I still have not found a solution, any help would be appreciated. Thank You

Aleishus, 2023-05-06:
@Taaaserfaaace Yes ，check the extruder motor  wires。

gillesguillemin, 2023-06-03:
I'm facing the exact same issue with a brand new EBB SB2209. I've ruled out any wiring issue and went as far as replacing the extruder stepper motor with a brand new one. 

No matter what, I'm getting this `TMC 'extruder' reports error: GSTAT: 00000001 reset=1(Reset)`

Feels increasingly like a driver issue but since I am not the only one it might also be software related, but what?

dkeekstra, 2023-08-22:
same here tried different currents but nothing seems to help

dkeekstra, 2023-08-29:
Spoke to a collegae of mine he said that it could be becouse the switching between StealthChop and SpreadCycle and turing one off would help

jusii, 2023-09-15:
Joining the club. Everything worked just fine for 2 months. And now I get this error almost everytime. Gone through all threads and discord. Nothing has helped.

Anyone found a solution?

jusii, 2023-09-16:
Guess I solved my problem. I think my problem was that filament movement was little restricted in the feeding bowden tube from spool. I attached it with zip ties to canbus cable awhile ago and looks like it restricted filament moving a little.

And then this maybe caused extruder driver on the sb2209 to overheat and make it reset.

akashrpatel, 2023-09-27:
I was having undervolt with the SB2240, so I switched to SB2209l, which I heard was not having the undervolt issue, but now on my first print with the SB2209, I ran into this issue...

jusii, 2023-09-28:
Small update from here. Ever since I opened the SB side door and have it just open during print, not one extruder reset. And it's been printing fine almost 2 weeks.

akashrpatel, 2023-09-28:
Hmm...so maybe overheating? I'll give it a shot thanks!

> Small update from here. Ever since I opened the SB side door and have it just open during print, not one extruder reset. And it's been printing fine almost 2 weeks.

jusii, 2023-09-28:
Weird thing is, I've been printing PLA so doors are open.

> Hmm...so maybe overheating? I'll give it a shot thanks!
> 
> > Small update from here. Ever since I opened the SB side door and have it just open during print, not one extruder reset. And it's been printing fine almost 2 weeks.

akashrpatel, 2023-09-28:
> Weird thing is, I've been printing PLA so doors are open.
> 
> > Hmm...so maybe overheating? I'll give it a shot thanks!
> > > Small update from here. Ever since I opened the SB side door and have it just open during print, not one extruder reset. And it's been printing fine almost 2 weeks.

I put this on there, without the fan, will see how it goes...

https://www.printables.com/model/574283-btt-ebb-sb22092240-voron-vented-door

I tired to switch back to my SB2240 and now it's worse than ever with the undervolt, before it was sporadic, now it's everytime...hope the 2209 works...very frustrating...

dryles, 2023-10-25:
I’ve been having this issue with my 2209 as well, recrimped my extruder motor connections as one seemed loose and it went away for a few weeks, now it’s back and I’m pulling my hair out.

akashrpatel, 2023-10-25:
I also re-crimped my wires and started to using the above door to allow ventilation - so far no issues. I've printed for a few weeks - knock on wood...

Goldon84, 2023-10-29:
I also started to have this issue after I switched to a sb2209 canvas toolhead. I checked the extruder motor wire and they looks fine. Since I was able to print without issue before using the cambia toolhead, I had a doubt about the on board TMC2209 stepper motor driver. So I rerouted 4 wires from extruder motor directly to my controller board, bypass the toolhead stepper motor in the printer.cfg, and now I can print without issue! Still don’t know what caused the issue: faulty connector on the toolhead board, or issues with the onboard stepper driver.

Borgratt, 2023-11-16:
I have recently also started getting this error. "TMC 'extruder' reports error: GSTAT:      00000001 reset=1(Reset)"
Seems to happen when i'm running long portions of my print that takes a lot of extrusion. Sounds like the extruder starts to skip before it throws the error like its not getting enough power.

ReadHoffman, 2024-01-15:
i;ve run thru multiple SB2209s, most recently I installed SB2209 2040 and still received this error constantly.  I tried every forum/discord/facebook fix i could find, turning off stealthchop, fixing umbilical movement, monitoring board temps.  None of this worked.  I finally replaced the motor with a different motor and rerouted the extruder motor wiring to go straight to octopus, not into SB2209.  This appears to have worked.  I was failing over and over again on the same print at roughly the same point and now it works.  We will see how it holds up.  Kind of defeats the point of CanBus tho.

cstorksen, 2024-02-05:
Hi, I also got this problem last week. Before that, it has been working fine for moths. I Have Stealthburner, EBB SB2209. I change the motor+ max curet,, upgraded klipper + firmware on SB2209 (and Octopus 1.1. board)but still the same problem. The printer crashes with the "TMC Extruder ....GSTAT:..." error after about one hour. Has anyone found a pollible solution or do I have to replace the SB2209?

jusii, 2024-02-05:
Since my september post, still printing without any problems. And I've been
printing A lot, 12 to 20 hours a day.

Could try closing the SB door and see what happens.

ReadHoffman, 2024-02-05:
As a followup to my reply above, I made a few mods that I suspect helped.  

I added a dedicated 5v psu and powered my pi off that. Seems to be helping canbus stability. 

Also, I did the piano wire thing that attached to the canbus wire. That has kept down the strain on the connector.

No failures since I did these2 mods. Still important to check can bus wire connections and not overheat the chips with an enclosure.

cstorksen, 2024-02-07:
Update to my "TMC Extruder ....GSTAT" error:
I have slowed my printer down, it got a little bit better, but still fail on big prints. I have quadruple checked the connection/plugg on the toolehead (and u2c), but still failing. I then connected extruder directly to the Octopus board -after that, I have not had any issues (printed for about 3-4 hour). This tels me that it is probably the driver on the EBB SB2209 board. I hope I can get a new board form BTT.

jcheger, 2024-03-16:
This issue is related to the SB2240 undervoltage issue (https://github.com/bigtreetech/EBB/issues/28). We seem to have the beginnings of a solution.

https://github.com/jcheger/BTT-SB22xx-undervoltage-reset-issue/blob/main/README.md

I did write a document describing all what I've done during debugging this issue, and why I came up with the solution of the motor grounding. We're all frustrated by this problem and the complete lack of support from BTT, and I feel it's important to leave a comprehensive report, so that everyone can benefit.

I hope it can help.

DVSVIDEO, 2024-03-17:
Unfortunately, I have suffered the same phenomenon.
A few months ago I converted from direct wiring to CAN bus, but I'm starting to regret it.
A 4 hour print stopped after 2 hours. New print and it stopped after 1 hour.
A lot of material and time wasted and I don't know the cause of the problem.
I will try this ESD thing by grounding...

nicknick2020, 2024-03-18:
> This issue is related to the SB2240 undervoltage issue (#28). We seem to have the beginnings of a solution.
>
> https://github.com/jcheger/BTT-SB22xx-undervoltage-reset-issue/blob/main/README.md
>
> I did [...]

After seeing your documented solution, I've ran an extra grounded wire to the toolhead and attached it to the motor casing via a screw + nut. So far I've completed a 4 hour print that failed the prior attempt. Here's to hoping this works to solve this (somewhat intermittent) SB2209 issue. :)

hoodwink55, 2024-04-14:
Been getting this error too on my EBB SB2209. I'm running it on SB with an LDO G2 extruder. Just like everyone else, was printing fine for months, then this started. The error is happening when printing PETG, w/ front doors open and top off. Even tried removing the SB wire cover... The error still happened. In reviewing my extruder config, I noticed that I had stealthchop on, so I set it to zero and everything seemed to be back to normal. A couple of long prints (Helldiver 2 helmet/35hr print) using PLA. Now I'm back to printing PETG and happening again. SB2209 board doesn't appear to be getting over 50c, but not sure where it's at when it fails cause I wasn't in the room.

stefanpieter, 2024-04-24:
> Been getting this error too on my EBB SB2209. I'm running it on SB with an LDO G2 extruder. Just like everyone else, was printing fine for months, then this started. The error is happening when [...]

I am having the exact same issue.

Xotron392, 2024-05-06:
I had this same issue recently. This is what I ended up doing. Replaced extruder motor, added ground wire from stepper case to earth, replaced SB2209(RP2040) with SB2209, replaced CAN cable, replaced power supply. However, none of this seemed to fix this issue also this was only happening when printing over 12hrs using ASA filament. Finally the fix that worked was adding a heatsink to the chip as well as a new electronics shroud with a 3010 fan onto my stealthburner. Something to note as well I lowered the bed temp by 10*C. Based on my issues it appears the root cause is the SB2209 board overheating. Before the fix EBB NTC temps would reach 75*C and the printer would throw the code then shut down. Since adding the heat sink, fan, and lowering bed temps. The EBB NTC temps max out at 59*C and I have not experiences any error regarding this since. I hope this helps someone. I was pulling my hair out with this.

kgrigio, 2024-05-20:
> I had this same issue recently. This is what I ended up doing. Replaced extruder motor, added ground wire from stepper case to earth, replaced SB2209(RP2040) with SB2209, replaced CAN cable, [...]

When you say you added a heat sink to the chip, I have a heat sink on the back side of the chip, but not on the chip itself, so do you have two heat sinks then or just one directly on the chip.  Also, do you disable stealthchop or still have stealthchop enabled?  Do you have a link to the shroud you used with the 3010 fan?

Xotron392, 2024-05-21:
> When you say you added a heat sink to the chip, I have a heat sink on the back side of the chip, but not on the chip itself, so do you have two heat sinks then or just one directly on the chip. [...]

I added a heat sink to the front of the chip. There is no room on the backside that i'm aware of to place one. I have stealthchop off. Here is a link to the shroud I'm currently using. I hope this helps. Cheer!  

https://www.printables.com/model/300272-sb2040-fan-cover

RimmeVG, 2024-06-11:
Hi,
I also had the error: "TMC 'extruder' reports error: GSTAT: 00000001 reset=1(Reset)" with the SB2209.
This error started when printing PETG, afterwards error with all filaments.
What solved it for me: I connected a ground wire from extruder motor chassis to mains ground (earth).
Printing for hours without problem now, also with PETG.

th0mpy, 2024-06-18:
Just to add my $0.02 here. I just started having the exact same problems with an SB2209. I just upgraded to the Gallileo 2, but prior to I had 0 issues with thousands of hours of prints with the standard clockwork 2. After the update to Gallileo 2 extruder I updated the distance per roration, gear ratio, and motor current; everything else remained the same and I transfered the board which was in the Clockwork into the Gallileo.

I would say that I have approximately 100 on the new setup and just today started to exhibit this behavior. I've tried several things with changing the motor currents, recrimping the motor connector, and updating to a slightly larger heatsink. Nothing seems to have worked. 

Now, I'm skeptical about grounding the motor, since it was not grounded when using the Clockwork setup. Fundamentally nothing has really changed so this really makes no sense to me as to why I need to ground the motor.  Has anyone made heads or tails of this yet?

AirMatress, 2024-06-24:
@DVSVIDEO Do you have a picture you could share of your solution?

th0mpy, 2024-06-27:
Ok, I'll admit that I might be wrong here. I took the suggestions and ran a thin wire from the planetary motor shaft over to the casing of the stepper. I then ran a grounding wire back to the chassis. 4-5 prints so far and no issues. 

I ordered some ABS ESD filament last night and printed the "[a]_rear_half_g2" and will be putting it in today. The theory here is that will allow the planetary shaft to ground through the bearing and out to the motor mounting screws, back into the wire connected to the chassis. It's the most eloquent way to deal with it that I could come up with, short of redesigning the entire system. 

I'll report back on my progress after the weekend.

th0mpy, 2024-06-27:
Side update. After further analysis I am printing both the front and back out of ESD filament. In the picture below it's the parts which are highlighted in yellow.

AirMatress, 2024-06-27:
Thanks for the update. Currently I'm thinking it's my can wire losing connection. I am able to print at 120mm/s but nothing higher. I read a bit about it [here](https://canbus.esoterical.online/troubleshooting/lost_communication_to_mcu.html) that convinced me more. GL!

th0mpy, 2024-06-28:
I thought maybe it was the same for me. I don't find any can bus errors in my logs though. Hopefully you're able to find out more. The best thing I can say with the canbus is to use the molded cable that they give you in the SB2209 kit, it's a much better connection than the one I made using the connectors from AliExpress.

AirMatress, 2024-06-28:
Like this one? https://www.aliexpress.com/item/1005004819126687.html? Mine was a cheapy that came with the Aliexpress kit.

th0mpy, 2024-06-28:
Either should work, however, I like the BTT one in your second link. It's much more rugged. The ones that caused me problems didn't have the over-molded rubber so it was just the connector with heatshrink tubing. It didn't seem to seat in tight.

th0mpy, 2024-06-29:
UPDATE. I've run two jobs of about 4 hours each since moving over to the ESD parts of the extruder. Again, the front and back of the Galileo are made from ESD-ABS with a grounding wire from the chassis of the motor housing. 
[image]

It's impossible to tell from the picture, but here you can see my ground wire.

Everything has gone well so far and no resets since. 

I'm not sure what else could be done from a design perspective to prevent the static build-up, but this is a viable option (albeit rather expensive).

AirMatress, 2024-07-01:
Thanks for the pictures will give it a shot!

th0mpy, 2024-07-01:
Keep in mind the ground cable is only part of the equation, you still need to dissipate the static build up from the planetary shaft inside.

AirMatress, 2024-07-09:
I found my problem (I think) after swapping out components and checking cables. It seems to be my extruder motor connector, which I thought I checked well but I did not. One of the crimped wires came off with a small tug. I am still doing testing to see if it appears again but for the first 3 prints it seems to be back to normal.

bigtreetech, 2024-07-30:
@Aleishus @MKryger1982 @Taaaserfaaace @gillesguillemin @dkeekstra   @akashrpatel @jusii @dryles @Goldon84 @Borgratt 
@ReadHoffman @cstorksen @jcheger @DVSVIDEO @nicknick2020 @hoodwink55 @stefanpieter @Xotron392 @kgrigio @RimmeVG  @th0mpy @AirMatress
"Dear Users. 

We are sorry to hear of the EBB reset issues you have been experiencing. After much investigation and research we have found that these issues are related to ESD events which result in discharge into the TMC driver. Here is what happens:

- Materials will do one of three things when rubbed against each another material: a. It will donate electrons to the other material and become positive charged. b. It will absorb electrons from the other material and become negatively charged. c. It will neither donate nor absorb since the electron bonds are too tight to permit this.
- The smoother the materials, the more surface area is in contact and therefore the more charge is exchanged. Hint, PTFE is very smooth.

To understand why this happened so much with PLA and not really at all with ABS we must understand how PLA, ABS and PTFE all interact with each other. The triboelectric series ranks materials according to their tendency to either accept or donate electrons. Materials that sit near 0 in the series will struggle to either donate or accept electrons and therefore they will struggle to develop a static charge. Materials that sit far away from 0 can develop significant charges. Let's see where ABS sits on the series (https://www.alphalabinc.com/triboelectric-series/):
[image]
With PTFE sitting at -190 and PU sitting at +60 you can see that ABS is actually very neutral. It donates a small number of electrons but not really enough to allow it to build up a significant static charge. Therefore, as it interacts with the PTFE, there is no significant static build up.

What happens with PLA? Let's look at the same series for how PLA and PTFE interact (https://ars.els-cdn.com/content/image/1-s2.0-S2211285518306335-fx1_lrg.jpg)
[image]
Here you can see that PLA and PTFE sit on complete opposite ends of the scale. Essentially they are a worst case scenario when it comes to building static charge. This means that the PLA will develop a massively positive charge over time as it interacts with the smooth PTFE. Since the gear from the extruder touches the PLA, this charge will be imparted to the gear. In turn, that charge will be imparted to the motor shaft, rotor and motor body which are all electrically connected via bearings. Eventually, the charge will be so large that it will jump from the rotor to the stator coils and pass into the TMC driver. While the driver can survive the hit thanks to the built in ESD protection, a reset event will still take place. The TMC2240 is more prone to resets during ESD discharge events and this is why the EBB SB2240 has a disproportionate number of users who have experienced this.

There is no real silver bullet that can prevent this on the PCB. Additional ESD protection can help to dissipate the ESD event but that protection will likely only start to conduct meaningfully after a voltage level has been reached that would cause a TMC reset anyways. TMC have therefore recommended that all users ground their motor body so that charge is immediately sent back to GND as soon as it builds up. There has been some discussion about whether PE or GND is better and the reality is that both are just as effective. The technical reasoning on this point is beyond the scope of this message.

The only effective solution is therefore to ground the body of your motor. To help users to do this, BIGTREETECH will be including a small grounding kit with each EBB sold in the future. It will consist of a ring terminal, an M3 x 35mm screw, 2 x M3 nuts and a small piece of wire. This is enough to create a clean path from the motor body to a GND pin on the EBB. We will also be updating our build guides to show users how to create an effective ground connection to the motor body.

While this is not a product fault, we do understand that you as users have had an experience with the product that could leave you unhappy because of this phenomenon. We would therefore like to offer to send you this small grounding kit to help you solve the issue. If you would like to receive a kit then please reach out to service003@biqu3d.com. The subject line of the email should say "TMC ESD issue" and the body of the email should contain: The store that you bought the product from, the order number, a screenshot of the invoice, your full name, your mailing address and a contact number for the delivery company.

Thanks for your patience while we investigated this phenomenon.
Sincerely,
The BIGTREETECH team."

robustini, 2024-08-17:
I don't know how you solved it, I can't in every way.
Re-wired the 'extruder' moons, ground its case, well cooled drivers, randomly I keep throwing away prints:

TMC 'extruder' reports error: GSTAT: 00000001 reset=1(Reset)

I am considering two things: to put the nozzle to GND (ground) as well, or to take a trip to Lourdes.

robustini, 2024-08-17:
Herer BTT SKR Pro with TMC5160, i don't know how you solved it, I can't in every way.
Re-wired the 'extruder' Moons motor, ground its case, well cooled drivers, randomly I keep throwing away prints:

TMC 'extruder' reports error: GSTAT: 00000001 reset=1(Reset)

And if the stepper were to lose a few microsteps on the extruder maybe on the print you wouldn't see anything at all, could it be that there is no provision for “just give a damn and keep printing”?
No, it is not provided, damn.
I am considering two things: to put also the nozzle to GND (ground) as well, or to take a trip to Lourdes.

bigtreetech, 2024-08-17:
> Here BTT SKR Pro with TMC5160, i don't know how you solved it, I can't in every way. Re-wired the 'extruder' Moons motor, ground its case, well cooled drivers, swap the driver, use another stepper position on the board but randomly I keep throwing away prints:
> 
> TMC 'extruder' reports error: GSTAT: 00000001 reset=1(Reset)
> 
> And if the stepper were to lose a few microsteps on the extruder maybe on the print you wouldn't see anything at all, could it be that there is no provision for “just give a damn and keep printing”? No, it is not provided, damn. I am considering two things: to put also the nozzle to GND (ground) as well, or to take a trip to Lourdes.

Are you speaking about a reset on the extruder toolboard? I am wondering because you also mention your motherboard and drivers. Grounding your nozzle could very likely also help. The hotend can develop massive amounts of static during a print and it is always good practice to ground it.

robustini, 2024-08-17:
> Are you speaking about a reset on the extruder toolboard? I am wondering because you also mention your motherboard and drivers. Grounding your nozzle could very likely also help. The hotend can develop massive amounts of static during a print and it is always good practice to ground it.

The reset is done by Klipper seeing that the TMC5160 can no longer communicate properly with the stepper with good probability, the reason I have yet to identify.
The only thing I have not changed is the motor, I will try that too.

bigtreetech, 2024-08-17:
> The reset is done by Klipper seeing that the TMC5160 can no longer communicate properly with the stepper with good probability, the reason I have yet to identify. The only thing I have not changed [...]

I see. This thread is specifically dealing with the toolboard. If you would like to start another thread on the SKR Pro or TMC driver github then it would be better as your comments here will be confusing to others who come to look for comments specifically related to toolboards in the EBB range. ESD strikes with an extruder motor connected to a mainboard rarely ever result in TMC resets since the resistance of the connection leads is enough to dissipate a lot of the energy so it is possible that your issue is not related to this at all.

Benoit-LdL, 2024-09-16:
> @Aleishus @MKryger1982 @Taaaserfaaace @gillesguillemin @dkeekstra @akashrpatel @jusii @dryles @Goldon84 @Borgratt @ReadHoffman @cstorksen @jcheger @DVSVIDEO @nicknick2020 @hoodwink55 @stefanpieter @Xotron392 @kgrigio @RimmeVG @th0mpy @AirMatress "Dear Users.
> 
> We are sorry to hear of the EBB reset issues you have been experiencing. After much investigation and research we have found that these issues are related to ESD events which result in discharge into the TMC driver. Here is what happens:
> 
> * Materials will do one of three things when rubbed against each another material: a. It will donate electrons to the other material and become positive charged. b. It will absorb electrons from the other material and become negatively charged. c. It will neither donate nor absorb since the electron bonds are too tight to permit this.
> * The smoother the materials, the more surface area is in contact and therefore the more charge is exchanged. Hint, PTFE is very smooth.
> 
> To understand why this happened so much with PLA and not really at all with ABS we must understand how PLA, ABS and PTFE all interact with each other. The triboelectric series ranks materials according to their tendency to either accept or donate electrons. Materials that sit near 0 in the series will struggle to either donate or accept electrons and therefore they will struggle to develop a static charge. Materials that sit far away from 0 can develop significant charges. Let's see where ABS sits on the series (https://www.alphalabinc.com/triboelectric-series/): ![图片1](https://private-user-images.githubusercontent.com/38851044/353415498-7a042e54-5937-4285-b97f-04f91384588f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY0Nzk5NDcsIm5iZiI6MTcyNjQ3OTY0NywicGF0aCI6Ii8zODg1MTA0NC8zNTM0MTU0OTgtN2EwNDJlNTQtNTkzNy00Mjg1LWI5N2YtMDRmOTEzODQ1ODhmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE2VDA5NDA0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg1MGU5YmE5MjNmMjdkNTc1NjQ5YjJiODhkMjZkZWVkYWM3OTEzOGYyYTRmODMyOTdhYTlmMTIyNWIyYTI2NjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GPX6nZ_dHSr1-BQ582wxsvIluwfZuXbTA0aI3nIdQxM) With PTFE sitting at -190 and PU sitting at +60 you can see that ABS is actually very neutral. It donates a small number of electrons but not really enough to allow it to build up a significant static charge. Therefore, as it interacts with the PTFE, there is no significant static build up.
> 
> What happens with PLA? Let's look at the same series for how PLA and PTFE interact (https://ars.els-cdn.com/content/image/1-s2.0-S2211285518306335-fx1_lrg.jpg) ![图片2](https://private-user-images.githubusercontent.com/38851044/353415808-087a94fb-5042-4cdb-bad9-14fca05e8d3f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY0Nzk5NDcsIm5iZiI6MTcyNjQ3OTY0NywicGF0aCI6Ii8zODg1MTA0NC8zNTM0MTU4MDgtMDg3YTk0ZmItNTA0Mi00Y2RiLWJhZDktMTRmY2EwNWU4ZDNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE2VDA5NDA0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzNjM2ODA4YzJkNzlkNWNkNDI1YmZmMDcyNjNkYWYwZmUzNTJjOTQ0YjM0OGJiNTc3YTA5MmU5NmFhMzY0YWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qQWfLavOGNgckVR8uuv9ALhh9FsM2uFV0_sFjRA4jVA) Here you can see that PLA and PTFE sit on complete opposite ends of the scale. Essentially they are a worst case scenario when it comes to building static charge. This means that the PLA will develop a massively positive charge over time as it interacts with the smooth PTFE. Since the gear from the extruder touches the PLA, this charge will be imparted to the gear. In turn, that charge will be imparted to the motor shaft, rotor and motor body which are all electrically connected via bearings. Eventually, the charge will be so large that it will jump from the rotor to the stator coils and pass into the TMC driver. While the driver can survive the hit thanks to the built in ESD protection, a reset event will still take place. The TMC2240 is more prone to resets during ESD discharge events and this is why the EBB SB2240 has a disproportionate number of users who have experienced this.
> 
> There is no real silver bullet that can prevent this on the PCB. Additional ESD protection can help to dissipate the ESD event but that protection will likely only start to conduct meaningfully after a voltage level has been reached that would cause a TMC reset anyways. TMC have therefore recommended that all users ground their motor body so that charge is immediately sent back to GND as soon as it builds up. There has been some discussion about whether PE or GND is better and the reality is that both are just as effective. The technical reasoning on this point is beyond the scope of this message.
> 
> The only effective solution is therefore to ground the body of your motor. To help users to do this, BIGTREETECH will be including a small grounding kit with each EBB sold in the future. It will consist of a ring terminal, an M3 x 35mm screw, 2 x M3 nuts and a small piece of wire. This is enough to create a clean path [... cut ...] t from, the order number, a screenshot of the invoice, your full name, your mailing address and a contact number for the delivery company.
> 
> Thanks for your patience while we investigated this phenomenon. Sincerely, The BIGTREETECH team."

I've had the same issue and instead of connecting the motor case to the GND of the EBB, I connected it directly to the ground of the powersupply. However I'm still having the reset issue. Is there a difference between connecting to GND or the actual ground?

th0mpy, 2024-09-21:
The only way I was able to address the issue was to print some of the enclosure with anti-static ABS. The issue with grounding the casing of the stepper is that there is no electrical connection between it and the shaft. Being that we're using a metal planetary gear, which typically spins around in an ABS envelope, you're only discharging static from a small portion of the equation if you're grounding the stepper body.  With the anti-static filament components, you're discharging the entire assembly, and since you're mounting the stepper on the anti-static components, you'll then discharge through the chassis. I've been running with the setup as noted above since June and I have hundreds of hours with no issue now.

Benoit-LdL, 2024-09-23:
Hi, as I was aware of the fact that the shaft isn't physically connected to the motor enclosure, I added a custom cable touching the shaft to ground it as well. So now both the motor enclosure and the shaft are connected to the ground. @th0mpy could you tell me if you connected it to the actual ground of the PSU or to the GND on the EBB board?

bigtreetech, 2024-09-23:
> Hi, as I was aware of the fact that the shaft isn't physically connected to the motor enclosure, I added a custom cable touching the shaft to ground it as well. So now both the motor enclosure and [...]

There is definitely a connection between the shaft and the enclosure. The shaft is supported on either end by metal ball bearings which in turn rest within the enclosure. Through those bearings there is a direct electrical connection.

There is no electrical connection between the coils and the body or shaft and this is where the issue arises. The body/shaft gradually accumulate charge and eventually it leaps from the body and into the coils and down the coils and into the motor driver which resets the TMC driver. Grounding the body will provide a discharge path that prevents build up.

Bear in mind that charge may well accumulate on the hotend too since the charged filament is flowing through it constantly. It is also therefore a good idea to find a way to ground the hotend if you continue to have issues as an ESD event conducted through the thermistor input could still result in ground bounce that resets the driver.

th0mpy, 2024-09-23:
I think there is a small misunderstanding. You are correct, the shaft of the stepper is electrically connected to the casing of the stepper through the bearings. The issue, however, is that everyone here is using the G2 extruder which has an aluminum planetary shaft with plastic gears, which electrically isolate it from the stepper.

The ESD filament solves this by grounding the planetary shaft through its bearings and back to the stepper casing.

I haven't experienced any ESD buildup in the hotend, but I think I'd solve it in much the same way. 

In my case I ran the grounding wire to the frame of the printer. It in turn is grounded to the outlet, since your frame should be grounded already.

bigtreetech, 2024-09-23:
> I think there is a small misunderstanding. You are correct, the shaft of the stepper is electrically connected to the casing of the stepper through the bearings. The issue, however, is that everyone [...]

Got you. Yes, that would be an issue that would result in buildup on the ABS body that would eventually leap into the motor/hotend/probe. Provided you can find a conductive filament that is not too brittle, using conductive on the tool head for most parts and then grounding the tool head would be the ideal solution.

seapear, 2024-09-28:
I've grounded my extruder motor case, and I'm still getting the error, what should I do now?

th0mpy, 2024-09-28:
Need more information. What is your setup (i.e. extruder, toolboard, etc)

seapear, 2024-09-29:
I have the sb2209 toolboard, using the clockwork 2 extruder and a rapido hotend

th0mpy, 2024-09-29:
Is the ground from your toolboard in fact grounded to earth?

seapear, 2024-09-30:
I soldered the cable to the main ground that connects straight to the canbus cable, which is definitely grounded

th0mpy, 2024-10-01:
The only other thing I can think of in terms of static discharge, have you grounded the hot-end as well?

[51 more replies]