bigtreetech/EBB #48: Document the use of 3 pin PWM + Tach fans with EBB36

sammcj, 2023-04-24:
It would be great if the use of 3 (or even 4) pin PWM + Tachometer fans could be documented.

For example - is it "safe" to use the PA0 (FAN0/1 header) for +/- and a single pin from the I2C header, say PB4 for the sense/pulse yellow wire? Are there any risks?

[image]

For context - It's generally a good idea to use a fan with a tach for your hotend cooling to avoid potential disasters should it fail or become blocked during a print - for example: https://ellis3dp.com/Print-Tuning-Guide/articles/useful_macros/hotend_fan_monitoring.html

longshanks1977, 2024-01-10:
Did you ever get an answer as to whether this setup is safe?

sammcj, 2024-01-10:
Nope, BTT has never responded to me :(

Igor-Yavych, 2025-03-12:
I used it like this 

```
[fan_generic test]
pin: !t1:PA0
max_power: 1.0
tachometer_pin: ^t1:PB4
tachometer_ppr: 2 
tachometer_poll_interval: 0.00040 #this needs to be set for your fan
cycle_time: 0.00004 #this needs to be set for your fan
off_below: 0.20 #this needs to be set for your fan
```

[image]

rennwaffen, 2025-03-13:
> I used it like this
>
> ```
> [fan_generic test]
> pin: !t1:PA0
> max_power: 1.0
> tachometer_pin: ^t1:PB4
> tachometer_ppr: 2
> tachometer_poll_interval: 0.00040 #this needs to be set for your fan
> cycle_time: [...]

Any particular reason to use an ST16 diode?

Igor-Yavych, 2025-03-19:
> Any particular reason to use an SF16 diode?

I just had it available. You probably should use something like 1N4148 instead. I have updated the image to reflect that.

oldmate45, 2025-06-10:
Just in case anyone else comes across this.
I can confirm that a BAT-85 will do the job just nicely as well, you might have some laying around if you used an Omron sensor.
In my case I used a 3 pin fan (Delta 5V) with PB9 as the tach pin, 5V from the probe and PA1 as the ground.

[image]