bigtreetech/EBB #49: SB2240 manual specifies incorrect connectors

gx1400, 2023-05-01:
The SB2240 Build manual specifies GH1.25 connectors for many of the board side connectors, like the TH0 thermocouple connector, etc.

GH1.25 connectors do NOT mate.

Discord notes that they are Molex Picoblade compatible connectors (preterminated wire and connector kits can be found on amazon US with an 'MX' designator')

Please update manual accordingly

gx1400, 2023-05-01:
SB2240 3D Model Step file has the 4-pin MAX thermocouple connector as Molex 53047-0410
https://www.molex.com/en-us/products/part-detail/530470410

1.25mm Pitch, PicoBlade PCB Header, Single Row, Vertical, Through Hole, Tin Plating, Friction Lock, 4 Circuits, Tray, 2.30mm PC Tail Length

miracuruzrh, 2023-08-25:
Confirmed. Yes, that are Picoblade connectors. The wrong information in the manual had cost me hours of troubleshooting until someone on a discord channel told me this information. Please update the manual. Thank you.

RicRavGithub, 2023-09-23:
The documentation also doesn't specify that the CAN connector cable needs a Amass XT30(2+2) female type.
Without this information, it's not possible to self-build this cable.