bigtreetech/EBB #52: EBB 36/42 Connector Details, including their type, pitch and names...

aaronse, 2023-05-15:
@bigtreetech consider updating manual, pinout or schematic diagrams to describe connector type and pitch details.

For example, am unable to figure out which crimp inserts to use for the EBB36 power/CANBus connector?

The EBB36 4 pin CANBus connector seems to be 3mm pitch based on my cheap calipers, and a dimension inspection of the .step model loaded into Fusion 360.  Ideally, the manual/pinout in this repo would have these details.

CONN-TH_4P-P3.00_30002-04WA-T1P seems to be also known as Molex Micro Fit 3.0 for anyone else looking.
 
Multiple people have burned time on this very question, see https://www.reddit.com/r/BIGTREETECH/comments/12vhc55/4_pin_connector_on_ebb36/

0-M, 2023-12-10:
this was super helpful, thank you!

@bigtreetech please consider adding this to the docs

numindast, 2024-05-06:
Please update documentation site with this very useful information. Thank you!

jjl40, 2025-02-12:
I could not find anything worthwhile on CONN-TH_4P-P3.00_30002-04WA-T1P. However, Digikey has many varieties of 4 female socket connectors. Would be nice to know which part number to use.
Well, BTT where are you???

aaronse, 2025-02-12:
@bigtreetech you're welcome to use my image and info for free, however you want.  I relinquish any and all copyright claims.  Please help your Customers be successful with their projects.

Credit my @AzaB2C printer/CNC/hack/make stuff channel if you want, free stuff is welcome or course.  Lol, if you don't ask, you don't get 😁

nils-hesse-xelera, 2025-03-05:
I think the power connector is this one: https://www.molex.com/en-us/products/part-detail/430250400
Not confirmed by trying it though. The part packed is from JMCONN.

irrenhaus, 2025-03-05:
> I could not find anything worthwhile on CONN-TH_4P-P3.00_30002-04WA-T1P. However, Digikey has many varieties of 4 female socket connectors. All of which are Molex Micro-Fit. Would be nice to know which part number to use. Well, BTT where are you???

I think the power connector is this one: https://www.molex.com/en-us/products/part-detail/430250400
Not confirmed by trying it though. The part packed is from JMCONN.

TheMightyGinger4, 2025-09-13:
@bigtreetech I'm here because I'm having this problem, and there is no guidance to be found anywhere. Why did you guys mix connector types, use uncommon connectors, and then document NOTHING about them except to say "we use JST connectors!" when you clearly also use microfit connectors in this product?

I am extremely disappointed and I might wind up returning this board because I have lost most of a Saturday afternoon TRYING to chase down information about an esoteric connector that you didn't bother to document. Please fulfill this request that has been open for TWO YEARS NOW. Thanks.