bigtreetech/EBB #56: Please provide STEP files for custom CW2 main body

garyd9, 2023-06-05:
Please provide the .STEP files for the two StealthBurner/CW2 modified files in "EBB SB2240_2209 CAN/Custom Printed Parts/"

Specifically, cable_door_EBB_SB.stl and main_body_EBB_SB.stl.  I typically use a CW2 mainbody that's modified to accept an ECAS (bowden collet.)  This requires an additional modification of the CW2 main body part.

ndanyluk, 2023-06-30:
@bigtreetech would it be possible to provide these STEP files like you do for the other custom printed parts? I am in the same position as @garyd9  and want to use this part but need to modify it to match the part it is replacing.

ndanyluk, 2023-07-04:
Whoops just realized these files came from a PR from @LobodPl not BTT themselves.  @LobodPl any chance you could open a PR with the step files for this modified part?  I want to use it but need to add threads to it for a PC4-M6 passthrough coupler (ERCF)

EduVZ, 2023-07-10:
@LobodPl , I'd also appreciate having the CAD files (STEP, F3D, IGES, almost anything is better than STL). I also just need to add a minor mod to enable the CW2 main body to accept a bowden collet.