bigtreetech/EBB #57: SB2209 not found on CANBus

FlorianWilk, 2023-06-07:
I can't get my SB2209 working over CAN, need help:

I have an Octopus MCU with Klipper, which is connected via CAN oder USB with my PI. 
"~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0" gives me the uuid of my Octopus MCU (no SB2209 connected at this time).

Then i flashed Klipper on the SB2209 (according to the Documentation, STM32G0B1, 8Mhz, PB0/PB1, 1000000Baud). Flashing went fine ("File downloaded succesfully") via USB. I removed the 5V jumper and set the CANBus Terminator Jumper R120 on the SB2209, connected it to the Octopus via RJ11.

But it's not found on the CanBus, no matter what i try (flashed several times now, checked wiring of the RJ11, etc..).
"~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0" still gives me only one uuid - the one from my OctopusBoard

LEDs on the SB2209 are both on (green and red). Not sure what this means, can't find it on the documentation.

Would be cool to get it up and running, but it's pretty frustrating right now.

BladerunnerxRC, 2023-09-05:
I have had at least 3 issues where the usb cable was sub par or may have been just not good enough. Also make sure it is not a power only usb cable.

viniavila, 2024-08-05:
I'm having the same problem. I'm using the PD0/PD1 pins in mcu klipper for CANBUS port (RJ11 connector in Octopus Pro) and PB0/PB1 in sb2209 klipper for CANBUS (as indicated [
[image]
](url). Flashing runs smothly. But when I run ~/klipper/scripts/canbus_query.py can0, just mcu can_uuid shows. In sb2209, the red led blinks and the green led was steady. Is it possible to be a cable issue? I'm using BTT original cable, that comes with the board

Goshaman, 2024-08-11:
Im having the same problem with my Siboor 2.4 kit, any chance you got it fixed with a new cable?

TerryFrench, 2024-08-13:
The only thing that worked for me was not a new cable, but a new board. I suspect that some component or solder was defective with the previous one, as nothing I did worked. 
New board, same flashing, and all worked.

Goshaman, 2024-08-19:
A new CAN board or main board?

TerryFrench, 2024-08-20:
CAN board (to put in the tool head). 
I bought and tried some other CAN boards to communicate with it, and it was not a problem with my main board or wires, but something on the tool head CAN board that could be flashed once but never answer then.

Goshaman, 2024-08-20:
I see, I will try the same, thank you!