bigtreetech/EBB #58: SB2240 Thermocouple issue

Tonykostansek, 2023-06-08:
Can anyone help me figure out why my EBB2240 is displaying  -93 degrees for my Hotend temp?  Im using a Rapido hotend and i think its Pt1000.  I have tried using these settings 
sensor_type:MAX31865
sensor_pin: EBBCan: PA4
spi_bus: spi1
rtd_nominal_r: 1000
rtd_reference_r: 430
rtd_num_of_wires: 2
If i try the setting below I get thermal couple issue. 
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: EBBCan: PA3
control: pid
i get -93 degrees. 
im connected 2 wires to the four pin connector and the jumpers are PT1000 1-on 2-on 3-off 4-on
please help. i even replaced this board and it still dont work.
tkostansek — Today at 10:00 AM
MCU 'EBBCan' shutdown: Thermocouple reader fault
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

arielbernal, 2023-08-25:
same thing here, did you get it working?

hallicarnassus, 2023-08-27:
I also have the same issue... 
However I tried setting the SPI section but it kicked back errors.

My settings are simply (with the SPI section commented out):
```
[temperature_sensor EBB_NTC]
sensor_type: PT1000
sensor_pin: EBBCan: PA4
# spi_bus: spi1
# rtd_nominal_r: 1000
# rtd_reference_r: 4300
# rtd_num_of_wires: 2
```

Can anyone offer a valid solution?

arielbernal, 2023-08-27:
I got it working using the following configuration. In my case one of the thermocouple wire broke when I was replacing the board. You can use a multimeter to check the resistance. It's a bit tricky with that tiny connector though. The MAX31865 is way way more accurate than the standard pin PA3

`sensor_type:MAX31865
sensor_pin: EBBCan: PA4
spi_bus: spi1
rtd_nominal_r: 1000
rtd_reference_r: 4300
rtd_num_of_wires: 2`

I'm supper happy with the SB2240 it never goes above 65 deg even without a fan, and with the enclosure at 50 deg, and the bed at 100 deg!!
I bought a SB2240 because one of the resistors of my fly-sb2040 burned because it overheated (using a small fan on the SB cover). All my problems got solved with the SB2240, the logo neopixel LEDs are not flickering anymore, the temperature is lot more stable and the 2240 driver is great. I didn't want to send the 4 cables of the extruder with the CAN cable. The extruder motor runs great with 0.4 amps. I can't believe I wasted soo much time with the fly-sb2040 

`[tmc2240 extruder]
cs_pin: EBBCan: PA15
spi_software_sclk_pin: EBBCan: PB10
spi_software_mosi_pin: EBBCan: PB11
spi_software_miso_pin: EBBCan: PB2
driver_TPFD: 0
run_current: 0.4`

hallicarnassus, 2023-08-30:
Thanks for the advice @arielbernal I've integrated the change into my config, measured the resistance of my PT1000 (was pretty much bang on 1000 ohms) however now klipper is reporting 

`octoprint.plugins.klipper - ERROR -  MCU 'EBBCan' shutdown: Thermocouple reader fault`

arielbernal, 2023-08-31:
I assume that's a 2 wire PT1000 right?
A couple of things:
- look at the tiny pins, probably use some stripped copper cable to make the connection with the multimeter. Also check that the pins make contact with the board male connector. I've seen the cables come off easily from the female connector.
- The message seems to be related to the connection. You can also look at the /print_data/logs/klippy.log to see if there's an invalid value or if that's the only error.

I remember having exactly the same error, so most probably is the connector or the cable. I think there's a new version with JST PH or HT connectors. How much space did they save by using those tiny MX connectors? I don't even have a crimper for those. I use the cables that came with the board.

hallicarnassus, 2023-09-10:
Thanks for the suggestion @arielbernal, it is a two wire PT1000 I have checked the resistance and connection to the board, both are fine, I initially just crimped the pins directly to the PT1000 wires, which also were showing up as being connected (with a multimeter) as fine but had the same problem, then tested second one which I had made using a couple of the pre-crimped wires spliced to the PT1000 wires. Both were showing the same result. 
I believe that there may be a slight manufacturing issue with the board itself, I have read in a few places that there have been reports of dry solder joints in some of the connected components near the plug that have been the cause of the issue. 
I'll run a rework on the board with some flux and a hot-air station to see if that improves anything. 
Hoping not to have to go back to using a thermistor as I have some high temp material I'd like to start printing with in the near future.

JamesnWalsh, 2023-10-27:
How did you get on with this? I'm getting the exactly the same problem.

Tonykostansek, 2023-10-28:
I ultimately dropped the 2240 and went to the 2209.  It’s worked fine.

jusdino, 2024-03-24:
I ran into basically this same issue. This isn't mentioned anywhere in the SB2240 manual, but [this guy on YouTube](https://youtu.be/oGdIyosQuRk?si=0pV645pY8MMeG5e8&t=1208) explained that the SB2240 comes with a built-in MAX31865 for measuring your PT1000 over SPI. All that to say that your `sensor_type: MAX31865` is actually correct. I finally got mine running with a 2-wire PT1000 on the SB2240 with almost exactly what you had:
```
sensor_type: MAX31865
sensor_pin: PA4
spi_bus: spi1
rtd_nominal_r: 100
rtd_reference_r: 430
rtd_num_of_wires: 2
```

Note that your `rtd_nominal_r` has one more `0` than I have.