bigtreetech/EBB #59: EBB SB2209 Heater not working?

Arkadije, 2023-06-11:
I managed to configure everything using the C2U adapter. 
Everything works except the heater. 
It does not give any signs of life. I have two different hotels with different heaters, so I can for sure say that it is not the faulty heater.
I just replaced the port in my config for all other sensors, so I did the same thing for the heater. 
As I said, all other sensors, extruders, and end-stops are working only it does not want to start heating.
 
```
[extruder]
step_pin: EBBCan: PD0 #PB3
dir_pin: EBBCan: PD1 #PB4
enable_pin: !EBBCan: PD2 #!PD2
rotation_distance: 22.811396174424
gear_ratio: 50:10
microsteps: 16
full_steps_per_rotation: 200
nozzle_diameter: 0.400
filament_diameter: 1.75
heater_pin: EBBCan: PB13 #PC8
sensor_type: Generic 3950
sensor_pin: EBBCan: PA3 #PA0
min_temp: 10
max_temp: 280
max_power: 1.0
min_extrude_temp: 10
pressure_advance: 0.051835
max_extrude_only_distance: 400.0
```

italiann, 2023-09-30:
It is not working on my device as well

Arkadije, 2024-04-01:
Sadly now I haven't. While trying to figure out what the issue was, I let the magic smoke out, so I got a new one which was fine.