bigtreetech/EBB #61: PT1000/TH0 mislabeling/clarification?

shoikan, 2023-06-14:
On the circuitboard of the EBB36 v1.2 a 2-pin header left of the center, bottom row of connectors is labeled 'PT1000'. On the wiring/pinout diagram that header is labeled TH0. There's also a header to the far right, bottom row of connectors without any markings on the pcb, right below the bank of microswitches, that's labeled PT100/PT1000 on the diagram.

Can you provide clarification which is to be used for which situation, and the correct pinout for the 4-pin connector? I _think_ the PT1000 labeled 2-pin is for the emulated PT1000 sensor (like on the M5P board), and the unmarked 4-pin for the header that is connected to the MAX31865 chip, but it is not clear at all from the documentation.

Additionally, the pins to be used for a 2-pin sensor via the 4-pin  connector are not specified. From other sources on the web I think those need to be connected to the center 2 pins, but it would be nice to get clarification here.

fryc88, 2023-07-07:
Yep, i just got to that stage today and i've noticed there's no pins described in PT1000 connector...Pretty common with BTT and other china manufacturers manuals/pinouts, so looks like we need to guess

shoikan, 2023-07-07:
I have it figured out, it's just not very clear. The two pin header marked PT1000 is only necessary if you use the two pin thermistor header TH0 and have a PT1000 sensor connected to it. It then uses a software emulation to estimate the temperatures. It's apparently very inaccurate at low temperatures. This is explained on page 9 of the EBB36 english manual.

On page 10 there's a very summary drawing of the wiring of a 2 wire and 4 wire PT1000 sensors, when you've got an EBB that has got an MAX31865. Basically: if you have a 2 wire version, wire the mid two pins. Because it had no text next to it whatsoever I missed it on reading.

@bigtreetech the info is apparently in the doc (as I found it), but it could definitely do with some clarification.

fryc88, 2023-07-07:
I've already found it too, that i need to use two middle pins, but on diagram available right in their github you can see every connector described with pinout except of pt1000...

Fuergrissa1, 2023-09-17:
So which pin do you use with for 2 wire sensors on the 4-pin connector. eg PA1, etc.. this doesn't seem to be labeled anywhere but must be set in klipper config.

appleimperio, 2023-09-17:
@Fuergrissa1 are you using a Max amplifier?

Drew3164, 2023-11-23:
Does anyone know what I'm doing wrong I can get my 2 wire PT1000 to work with with the Max31865 EBB2209

 [extruder]
step_pin: EBBCan: PD0
dir_pin: !EBBCan: PD1
enable_pin: !EBBCan: PD2
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: EBBCan: PB13
#sensor_type: EPCOS 100K B57560G104F
#sensor_pin: EBBCan: PA3
sensor_type: MAX31865
sensor_pin: EBBCan: PA4
spi_bus: spi1
rtd_nominal_r: 1000
rtd_reference_r: 4300
rtd_num_of_wires: 2
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250