bigtreetech/EBB #62: Missing STL for canbus cable with difrerent shape

KianSajjadi, 2023-06-19:
Hey, I have the ebb sb2240 with a different cable shape than the one advertised (came in the box) and the STL doesn't fit around the connection, 

Cheers

lixibert, 2023-06-20:
Same here, the new cable doesn't fit to the custom printed part can adapter. It seems that the rubber sleeve is way to big for it.

Phil-x, 2023-06-22:
Same here with an sb2209 kit.

Phil-x, 2023-06-22:
``

kieraneglin, 2023-07-02:
Any update here?

kieraneglin, 2023-07-02:
It's not perfect, but I've uploaded an updated model [here](https://github.com/kieraneglin/Voron-Parts/tree/master/CAN%20Cable%20Holder)

Phil-x, 2023-07-02:
EBB published a new version last week : https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/Custom%20Printed%20Parts/Printed_Part_for_CAN_Cable_v1.1.stl.
The fitting was way better but, to me, far from ok. If ever you're interested I published a remix of this part here : https://www.thingiverse.com/thing:6105666

kieraneglin, 2023-07-03:
@Phil-x that didn't work for me because it interferes with the USB-C plug - I'm willing to bet that model was made for the variant that doesn't include USB

Phil-x, 2023-07-03:
Yep. That's why I did a remix. I shaved 2 mm from the bottom to allow for the USB plug.

Phil-x, 2023-07-20:
Hi,

Step file has been added to the thingiverse https://www.thingiverse.com/thing:6105666
:-)

Phil.

KianSajjadi, 2023-08-16:
Cheers for the solutions, yeah I noticed that the USB-C was interfering when I printed the original and the remixed one worked well

bennierex, 2023-10-11:
Same issue here, no updated part from @bigtreetech I see...

JKtech3D, 2023-11-23:
Found this: https://github.com/MagicPhoenix/MPX-VORON-24R2-CBT/blob/main/Printed%20Parts/Toolhead_Can_Cable_Fix.stl