bigtreetech/EBB #64: Pinout for SB000 missing, no fan is working

diberben, 2023-07-17:
Hi all,
after connecting the SB2240 to my Voron Tap with LGX lite, no fans are working.
From my printer.cfg:

[fan]
pin: EBBCan: PA0

[heater_fan hotend_fan]
pin: EBBCan: PA1
heater: extruder
heater_temp: 45.0

Temperature reading is nominal (currently 55°C), dashboard indicates Hotend Fan 100%, but vfan1 and vfan2 on SB0000 do not output any voltage. Fan is a Sunon MF40102V2) and is working (checked with lab PSU).
Whats wrong?

OneHotTake, 2023-07-19:
I have the exact same problem.

Delanjp949, 2023-07-23:
so do I

Nemernemer, 2023-07-25:
This may be redundant, but please make sure you have commented out / removed the original [fan] and [heater_fan hotend_fan] sections in all configuration files.
Klipper will not warn you if you have these defined twice.

grimlokason, 2023-08-01:
Take care, it seems that you inverted the pin ( but you followed the manual :') ) : https://github.com/bigtreetech/EBB/issues/29

CharlotteBeLike, 2024-07-29:
BTT hasnt created a proper pinout for their EBB SB0000, so I did. Please share this so others can use it. 

[image]
[image]

bigtreetech, 2024-07-29:
> BTT hasnt created a proper pinout for their EBB SB0000, so I did. Please share this so others can use it.
> 

Build guide page 01.

[image]

Since the SB0000 is used with both STM and RP2040 parts, specifying the pin numbers on the drawing would result in confusion. Therefore only pin names are used and the specific cfg files for each toolboard have the relevant pin numbers included.