bigtreetech/EBB #65: main_body_EBB_SB now differs from new standard motor plate

Greecher, 2023-07-25:
About mid July they updated the Stealthburner STL's, and one of the changes was to the motor_plate.  The change is a modification to the large locating pin near the top of the body that fits with the main_body.  No other change is apparent.  This should be modified, as that feature cascades to the latch changes also.   I have created a one off stl, but best if the cad could be updated.  Also request that the step files for these be added.

avalonea, 2023-08-06:
So has it been addressed yet I'm finding out, I have the same issue.
@Greecher, don't suppose you could share your temp fix? I'm kind of jammed up by this issue.

bkt0031, 2023-08-06:
My plan of attack, for now, is to downloaded the original released version of the motor_plate STL from the VoronDesign Github site and print that one out to use with this main_body part. I loaded both versions of the motor_plate into SuperSlicer and the only difference that I could see is that cutout shape. In fact looking at the slice layers of the new motor_plate the original cutout shape is still there on layer 30 and then gets bridged over to the new shape on layer 31. Reading the change notes has not shed any light as to the reasons why in my mind.

Greecher, 2023-08-06:
Avolonea,
  here is the modified stl.  This worked well for me,

Best Regards,
Greecher

VovyCZ, 2023-08-10:
> Avolonea, here is the modified stl. This worked well for me, Best Regards, Greecher

Where is the modified stl please?

Z.

Greecher, 2023-08-10:
Lukas,
  the BTT github doesn't accept files, and I didn't note she wasn't cc'ed
in.  I had to create a git repository of my own, here is a link to that
file:
https://github.com/Greecher/Github-Repos/blob/main/main_body_EBB_SB%20ModH.3mf

Best Regards,
Dave aka Greecher.

greendog99, 2023-08-11:
I just ran into this today while trying to assemble my CW2 latch. The new, thicker latch doesn't fit.

@Greecher that URL gives a 404 error, probably because of the space in the filename. I assume the 3MF here is the correct file? https://github.com/Greecher/My-Voron-Mods/tree/main/V2/StealthBurner

Thanks!

Greecher, 2023-08-11:
Mike,
  yes, you are correct, grab the main_body_EBB Fix.3mf file.  I dumbly
modified the file name to make more sense, and doh!  Thanks for catch.
Here is the full and unchanging url to the fixed part:
https://github.com/Greecher/My-Voron-Mods/blob/main/V2/StealthBurner/main_body_EBB%20Fix.3mf

Best Regards,
Dave

Greecher, 2023-08-11:
All,
  also, here is the f3d file of the comparison.  There is the
original Stealthburner main body, which has the correct interface with the
motor plate, but lacks the thinner section where the door is.  There is the
original BTT main body (EBB), still in mesh form, and last is the fix which
started work from the BTT file, but did a nose job on the top pin that
interfaces with the motor plate.

Best Regards,
Greecher

Greecher, 2023-08-11:
And the file...
https://github.com/Greecher/My-Voron-Mods/blob/main/V2/StealthBurner/main_body_EBB_Comparison%20v1.f3d

appleimperio, 2023-09-10:
@Greecher After trying your fix the door is not closing properly. I print this door "cable_door_PCB_EBB_SB.stl" The door needs to be fixed too? Thanks

Greecher, 2023-09-10:
As far as I can see, the original files have been modified, and they
increased the cable cover width, and thinned the mating section on the main
body, thus the doors are too wide for the version of the mod I did.  That
looks like a good thing, and they only did it 2 weeks ago, but confusingly,
they didn't fix the original issue with the mating peg from main body to
motor plate.

As that the case, I did a quick mod to their main_body_EBB_SB files(step
and stl), which I could properly do with the step file provided.  So here
is the modified files, which work with the new door options
(cable_door_pcb_ebbv or cable_door_ebb).

https://github.com/Greecher/My-Voron-Mods/blob/main/V2/StealthBurner/main_body_EBB_SB_MOD_SX.stl
https://github.com/Greecher/My-Voron-Mods/blob/main/V2/StealthBurner/main_body_EBB_SB_MOD_SX%20v2.step

Note to appleimperio, just reprint the above, and use it instead of my
original mod, and with the cable door you already have.

Greecher

appleimperio, 2023-09-10:
@Greecher Thank you!!