bigtreetech/EBB #71: EBB36 v1.2 USB flash

Collapsednut, 2023-09-02:
I'm trying my hardest and searching I promise. 
I'm trying to flash via usb in dfu mode. I configured per the v1.2 manual. Device shows as STMicroelectronics STM Device in DFU Mode. I am not using canbus, I intend to use this in USB as a traditional mcu. 

Flash fails 
Flashing out/klipper.bin to 0483:df11
sudo dfu-util -d ,0483:df11 -R -a 0 -s 0x8000000:leave -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08000000, size = 28528
Download        [=========================] 100%        28528 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status

Failed to flash to 0483:df11: Error running dfu-util

MartijnGevaert, 2023-10-19:
I am getting this same issue. Not clear what to do here. Would like to flash the EBB36 from raspberry pi directly over USB

dotorg, 2023-11-20:
The failure is a red herring. The "File downloaded successfully" is all that matters. DFU Util fails right after that trying to get the status because the board has reset and isn't in DFU mode anymore.