bigtreetech/EBB #73: EBB2209 Communication issues resolved

sf-pfm265, 2023-09-11:
I was having a lot of trouble getting my EBB2209 setup to work.  Klipper was having trouble establishing communications over the CanBus. After 35 seconds (which is the timeout set), Klipper would shut down.  My son and I spent weeks on this issue before discovering the root cause. I want to report it here as I suspect others will run into it. If you are having communication issues you might want to check to see if you have this condition in your setup. It turns out that this problem is easily fixed.  

First, the problem manifests as randomly spaced communication errors. When I would run

`candump -ie can0,#FFFFFFFF`

We would see errors as 

`ERRORFRAME
        protocol-violation{{tx-dominant-bit-error}{}}
        error-counter-tx-rx{{235}{0}}`

When we attached a logic analyzer, we would see mal-formed packets that would occur as a result of sporadic voltage anomalies between CAN Hi and CAN Lo.

When I would remove the daughter card (the card that sits in the Stealthburner shroud and attaches to the EBB2209), the errors would go away.  

I'm using an LDO 5015 blower on my Stealthburner.  That fan has a metal foil label.  The daughter card has solder nibs that stick out the back of the card. These nibs make contact with the foil label.  When I removed the label and reattached the daughter card and shroud, all of the errors disappeared.  

I don't claim to have a full understanding of the mechanism at work, But I suspect that a static charge is building up on the label and once it reaches a certain threshold, the static discharges causing a voltage spike. If this is true, then it's also possible for a charge to develop on the plastic itself -- even without the foil label. I don't know what kind of plastic is used in these fans and I've not seen the issue since removing the foil label, but out of an abundance of caution, I think people should insulate the back of the daughter board. 

The simple fix is to remove the label from the fan. But for an extra measure, I ground down the solder nips on the daughter card (just enough so that they are no longer sharp) and applied Kapton tape over them.

jonwally, 2023-10-23:
You saved me! This was my exact issue, except it was constant from the beginning. I could not get past uploading firmware to the SB2209 board. But once I removed the StealthBurner face board, I could see the UUID. I even used a DVOM to check the CAN wires and measured the 60ohms as expected, so didn't think anything was wrong. But there must be a tiny short somewhere.

It's 1:30am now and I have work in the morning. Thank you again.

sf-pfm265, 2023-10-23:
@jonwally the trouble returned for me a few days ago. Clearly, it was not the metallic label that was causing the issue. But, in all cases, when I removed the daughter card, the trouble went away.  When I reconnected the daughter card, the trouble returned. I then removed the screws that attached the card to the housing, and plugged the daughter by itself, and the trouble went away.  When I put the screws back in, the trouble returned. I further isolated it to one of the two screws (the one farther from the header). 

I suspect that the copper ground plane and one of the signal traces of the PCB are exposed in the side wall of the screw hole in the PCB and the screw creates a short. This is likely either a design defect in the SB0000 or a manufacturing defect with this particular board. I've been running it without this screw for the past few days and have had no problems. 
[image]

jonwally, 2023-10-23:
@sf-pfm265  After I saw your post, I spent about half an hour further narrowing it down. I think I created a short when I filed down some of the thru-hole solder joints so the PCB would sit flat; but I was able to get it working.  I know plastic screws are a thing so I might just order some from DigiKey when I order other things. Common practice when PCB designing, is to only have ground planes around screw holes if there is any copper. If there is a trace and a ground plane, then the screw could create a short. Definitely a good idea, just poorly executed.