bigtreetech/EBB #74: SB2209 / SB2240 Won't go into DFU Mode

alexrizk1117, 2023-09-19:
I'm trying to install an sb2209 board using U2C wired into a pi. I'm trying to get the board into DFU mode but haven't had any success. All I see when running "lsusb" on my pi is:

- OpenMoko, Inc. stm32f445xx

- OpenMoko, Inc. Geschwister Schneider CAN adapter

Also, I'm trying to use the Boot + Reset method, but that doesn't seem to work. Is the SB2209 supposed to have a solid green and red light?

Also, when having the U2C board hooked up, should it be running power through the 24V PS, or should that only be powered through the pi in usb C?

Nooni99, 2023-10-18:
Have you found a solution @alexrizk1117 ? I am having the same issue except I see " Raspberry Pi RP2 Boot" when I try to go to DFU mode. Despite this, I am unable to enter dfu mode.

ga73, 2023-10-28:
@Nooni99 i'm having the same issue...have you worked this out ?

Nooni99, 2023-10-28:
@ga73 I have not found the exact solution and I just gave up on it because I wanted to get my printer working and just wired everything directly to my motherboard without canboot. However, I will do some experimenting once I have time. The next thing I am going to try which might work is using stm32 cube programmer to rewrite the sb2209. That might work and I suggest you try it. Let me know the results.

ga73, 2023-11-03:
Okay I got everything working last night and now up and running... I connected both the utc and rp2209 to 24v and ground instead of 5v usb power... The only other thing I did different was to get into dfu on the rp2209 I held down the boot button and then pressed the reset for maybe 3 second.. Let go of reset and then boot button... My can wires went to the screw terminals on the utc... Green was can low and yellow can high... I hope this helps someone

ga73, 2023-11-03:
Also make sure you remove the usb 5v jumper BEFORE connecting to 24v and that you have the 120r jumpers on both the utc and rp2209 board otherwise they won't communicate with each other

ga73, 2023-11-03:
Also this link was much better than just using the BTT guide alone...

https://github.com/Esoterical/voron_canbus

ga73, 2023-11-03:
Your welcome, I hope it helps, I'm not expert but feel free to leave another message here if you get stuck and I'll see what I did differently.. Good luck

ParabolicTech, 2024-02-09:
I am having the same issue. I hold the boot button and press the reset and it doesnt go into DFU mode. The device ID shows up with no description. This happens weather or not I try to enable DFU. I can plug it into my windows machine and it WILL bring up the bootloader folder if I do the button sequence. But for some odd reason connected to my Raspberry PI it doesnt function into DFU mode. Anyone else with this issue? Is there a way I can flash it in Windows?

Nooni99, 2024-02-09:
@ParabolicTech have you tried the solutions from above?

ParabolicTech, 2024-02-09:
> @ParabolicTech have you tried the solutions from above?

Yes, Thats why I posted. Ive been at this for hours in multiple places and cant figure out why I cant enable DFU. Diffrent known working cables - different (5v) power supplies, in fact I have my raspPI disconnected from my octopus board JUST to eliminate some of the other addresses showing up.

Nooni99, 2024-02-09:
@ParabolicTech hmmm. That's interesting. I'm not sure what's wrong. Those boards are quite difficult to get running anyways and I hope bigtreetech does another version that's easier to get working. In the mean time, I hope someone posts a solution that works for you. Best of luck!

ga73, 2024-02-09:
You mention you connected to 5v..have you tried connecting to 24v? This seemed to make a difference for me, also make sure your usb cable is a data and not just a usb power cable and finally do you have the r120 jumpers on both your canbus board AND your UTC (assuming this is what your using)?

I connected mine up fully and then flashed it, saying that if you just can't get it working try stm32cube software on windows as a last resort

xPakrikx, 2024-02-10:
same problem ... nothing helps. i see only Raspberry Pi RP2 Boot under lsusb.

LeighT616, 2024-02-17:
Hi i've just hit this issue... the board will not go into DFU mode.  I've tried various USB C cables and zero dice!

is there any resolution on this, i'm looking into the stm32cub software via windows but surely there has to be a better solution

ParabolicTech, 2024-02-17:
So after a lot of trials and tribulations I found out it DOES go into DFU mode by holding the reset button and clicking the boot button once. HOWEVER, the feedback that's listed online isn't correct. The board im using has the RP2040 chip on it so it wont list the description when it goes into DFU mode as listed on various instructions. The description is just blank. I was able to continue with the process and flashed the board, got the green light and the flashing red light and the rest of the listed setup worked fine.

rafal-molotkiewicz, 2024-03-28:
I have the same issue. My board only shows in two modes:

`Bus 001 Device 058: ID 1d50:614e OpenMoko, Inc. rp2040`
or
`Bus 001 Device 057: ID 2e8a:0003 Raspberry Pi RP2 Boot`

Neither one shows when I try `sudo dfu-util -l`

I tried holding boot and short pressing reset, long pressing reset, double and triple click reset.
I tried above swapping buttons.

I tried all this with 5V power, 24V power, and 24V power with connected CAN data lines.

Nothing made any difference. How can I proceed from here?

Nooni99, 2024-03-31:
Yeah, I'm still having the same issue with getting it in DFU mode. I have tried everything.

wes4560, 2024-04-30:
I have my sb2209 rb2040 hooked directly to the octopus using the rj connection I have unplugged everything used the 5v jumper and also tried the 24 v supply and I can't get it to go in DFU mode the light just stays green and I can't get it to show up at all in the ssh with the pi

Nooni99, 2024-04-30:
@wes4560 I found the solution. Since the chip is an RP2040, it will use boot mode. Your device should appear as "Raspberry Pi RP2 Boot" which is effectively the same as DFU mode. You can download firmware to it with no problem. Best of luck!

mprenter, 2024-06-15:
Same issue as everyone above, having a brand new board and unable to get it into DFU mode.  Can get the device to say "RP2 Boot" which some people say is good enough, but how do I install the firmware without using the dfu-util?

Nooni99, 2024-06-15:
@mprenter please don't make the misunderstanding I did. "RP2 Boot" is the "DFU mode" of this board. Since the board has a RP chip, it uses "RP2 Boot" instead. So follow all the steps you should follow and once you have "RP2 Boot" you're ready to move onto the next steps. You should be able to install the firmware the same you would any other chip. Let me know if you have more questions.

looxonline, 2024-06-15:
> Same issue as everyone above, having a brand new board and unable to get it into DFU mode. Can get the device to say "RP2 Boot" which some people say is good enough, but how do I install the firmware without using the dfu-util?

This is not an ST chip. It is an RP chip. You can't use dfu-util with it, you need to use the programming methods mentioned in the manual.

closed 2024-06-15 by looxonline

Destra-Slaybot, 2024-09-02:
> Have you found a solution @alexrizk1117 ? I am having the same issue except I see " Raspberry Pi RP2 Boot" when I try to go to DFU mode. Despite this, I am unable to enter dfu mode.

was this ever solved, i can't get past this one part

bigtreetech, 2024-09-02:
> > Have you found a solution @alexrizk1117 ? I am having the same issue except I see " Raspberry Pi RP2 Boot" when I try to go to DFU mode. Despite this, I am unable to enter dfu mode.
> 
> was this ever solved, i can't get past this one part

If you see this, then it is in DFU. It is an RP2040. The manual has the info you need.

lunchbox2883, 2024-10-13:
Same situation here... Not getting the DFU mode but i am getting the PI RP2 boot. but getting the same error as @Charleslotto above when pushing through with the flash

dahpuffy, 2026-03-15:
Hey @Jax1er, sorry my bad. I looked at your lsusb output again and realized why my fix isn't working for you.

​BTT actually makes two versions of your board: one with an RP2040 chip, and one with an STM32 chip. Your lsusb shows stm32g0b1xx, meaning you have the STM32 version. The .uf2 drag-and-drop method I posted is only for the RP2040.

​The good news is that seeing "OpenMoko, Inc." usually means your board already has Katapult or Klipper installed, so I think you should be able to skip the DFU entirely and just flash your firmware over CAN or USB using the Katapult script:
python3 ~/katapult/scripts/flashtool.py ...

I had some problems with the BOOT and RESET buttons, so if you are struggling to get you board into DFU mode with the RESET button, try powering the board off completely. Press and hold the BOOT button, turn the power back on, and then release BOOT. This worked for me at least.

I don't have your exact board, but my EBB36 uses the same STM32 chip. I'm definitely no expert, but I literally just spent hours back-and-forth with Gemini with a similar issue when setting up CAN between my Kraken, EBB36 and Eddy Duo. 
I don't know your exact hardware and wiring to give you the exact command, but I hope this points you in the right direction!