bigtreetech/EBB #76: EBB SB2209 CAN V1.0 (RP2040) + Voron Tap = Will not trigger endstop

Doogle03, 2023-09-24:
EBB SB2209 CAN V1.0 (RP2040) + Voron Tap, suddenly unable to get Klipper to trigger the z endstop.  Tried shorting gpio21 directly to gnd and no change within Klipper.  Also tried connecting to gpio22 and gpio6 but nothing will get Klipper to trigger the z endstop.  Tried a fresh install of Klipper from scratch, using up-to-date install, no change.

Checked with Voron discord, I am not the only user of SB2209(RP2040) that is having issues with Voron Tap.

Doogle03, 2023-09-24:
Issue most likely with Tap module; using v2 PCB with 5v only config but signal isn't being sent to the EBB board.  I swapped X and Z endstops in the config and a mechanical switch was able to trigger the Z but lifting toolhead didn't trigger X.

Doogle03, 2023-09-25:
Connected signal wire from 'malfunctioning' tap module directly to Octopus v1.1 board and it was able to receive trigger signal and Klipper was able to pick it up.  Tap module isn't malfunctioning, BTT SB2209 RP2040 will not recognize trigger signal that is compatible with other boards

bugalaws, 2023-10-03:
Hello, i got the same problem, do you can to get this work?

bugalaws, 2023-10-03:
In my sb2209 rp2040 i have other trouble too.. my rgb led don’t work too.. if i connect them at the mks robin i can t control the leds, but in the sb2209 module doesnt  work…

pakerfeldt, 2023-10-15:
I have the same issue with TAP probe not triggering. TAP PCB seems to be working just fine. I've been able to measure 200 Ohm and 1k Ohm over G / S pin when normal vs triggered.

pakerfeldt, 2023-10-16:
One person I talked to solved this by switching out the sensor pin in the EBB connector from GPIO21 to GPIO22 in the probe connector, and also adjusting the printer.cfg accordingly. This did not help for me unfortunately. However, I managed to work around this by connecting the TAP PCB to the endstop connector on the EBB board using a BAT85 diode.

[image]

looxonline, 2023-10-17:
> EBB SB2209 CAN V1.0 (RP2040) + Voron Tap, suddenly unable to get Klipper to trigger the z endstop. Tried shorting gpio21 directly to gnd and no change within Klipper. Also tried connecting to gpio22 [...]

Something does not add up with your testing. In this comment you mention that you were unable to get the z to trigger by manually shorting the z endstop whereas in the next comment you mention that you were able to get it to trigger by using the x limit switch. Electronically these processes are the same and should have the same result. It is therefore logical to conclude that there is something amis with your testing method or your config. For example, you should have a pull up enabled on your input pin since the TAP PCB is acting as a normally low buffer. Your other option is to bridge solder jumper 2 on the TAP PCB which would pull the pin high. You also need to ensure that your logic is inverted in the config since the pin is high when the tap is inactive and low when it is active.

looxonline, 2023-10-17:
> I have the same issue with TAP probe not triggering. TAP PCB seems to be working just fine. I've been able to measure 200 Ohm and 1k Ohm over G / S pin when normal vs triggered.

Triggered on the tap should produce a reading of a very high impedance since the open collector output would go completely open. Not triggered should measure close to 0 ohms since the output should be pulling straight to ground. If you are not getting those readings then the issue may lay there.

[image]

looxonline, 2023-10-17:
Another possibility is that your TAP PCB is putting out the maximum voltage of 0.4V when it is supposed to be in the low state (most would not do this). If this is the case then the endstop input may not trigger since it requires less than 0.4V to trigger (it can handle up to 0.2V). You would need to use the probe input instead and ensure that you manually bridge solder jumper 2 on the TAP PCB to pull the sig line up. You should also enable the pull up on GPIO21 by prefixing the pin with the ^ character in the config.

pakerfeldt, 2023-10-18:
I guess my problem was tied to the fact that my TAP PCB is giving 200 Ohm when not triggered. No idea why it does that if that isn't the intention. I got my probe from lab4450. Either way, I managed to solve my issue with a diode and the endstop connector. Maybe it would have worked with diode and probe connector as well (as you mentioned to me in a DM over Discord).

christian-nils, 2023-11-16:
I got it working by using GPIO22, GPIO21 would not set up a pull-up resistor. GPIO22 did.

Nrdgy, 2024-01-28:
For anyone who comes across this thread in the future, here is a small but possibly important note:

ChaoticLab uses the following color coding for their included cable:
Red: 5V
Black: Signal
Blue: Ground

I wanted to continue using the ChaoticLab cable and therefore had to change the connector. Unfortunately, I assumed the standard assignment, black as earth. As a result, a "triggered" signal was never generated. It was only after I looked at the [schematics](https://github.com/Chaoticlab/CNC-Tap-for-Voron/blob/master/Hardware/CNC%20Tap%20V2-SCH.pdf) of the VoronTap board again that I realized this.

2bbionic, 2024-04-02:
Had exactly the same issue - switching GND and SIGNAL wires did the trick for me (SB2240 with TAP v2 on M8P).
Only had to invert the signal im my printer.cfg
`pin: !EBBCan: gpio22`

regards,

2bbionic