bigtreetech/EBB #78: EBB SB2209 rp2040 - NEOPIXEL DOESNT WORK

bugalaws, 2023-10-03:
Hello, i have e ebb sb2240 rp2040 …all things work well , extruder, thermistor, fans etc..  only the neopixel doesnt work. I allready have tested the neopixel directly in my mks robin nano and works well..

Some times when start de printer at de first time the leds comes green but if i change the color or the intensity it powering off the leds and stay off until reboot the printer

I am using this setup:

[neopixel sb_leds] 
pin: EBBCan:gpio16
chain_count: 3 
color_order: GRBW

jjkrygsman, 2023-11-11:
Maybe try setting the initial colors:

#initial_RED: 0.0
#initial_GREEN: 0.0
#initial_BLUE: 0.0
#initial_WHITE: 0.0

It's been working fine for me.

jasonmacdonald, 2024-02-19:
Did oyu ever solve this. I'm having a similar issue of colors not working right and if I try to go pure white, everything just turns off.

Gardosen, 2024-02-25:
same issue here.

rgbs turn on with a bluish default color, but the entire config does not work. no control possible.
this is fucked up hardly in my opinion...

JKtech3D, 2024-02-25:
> same issue here.
> 
> rgbs turn on with a bluish default color, but the entire config does not work. no control possible. this is fucked up hardly in my opinion...

post your neopixel config

Gardosen, 2024-02-25:
> > same issue here.
> > rgbs turn on with a bluish default color, but the entire config does not work. no control possible. this is fucked up hardly in my opinion...
> 
> post your neopixel config

`[neopixel sb_leds]
pin: EBBCan:gpio16
chain_count: 3
color_order: GRB

[gcode_macro _sb_vars]
variable_colors: {
        'logo': { # Colors for logo states
            'busy': {'r': 0.4, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'cleaning': {'r': 0.0, 'g': 0.02, 'b': 0.5, 'w': 0.0},
            'calibrating_z': {'r': 0.8, 'g': 0., 'b': 0.35, 'w': 0.0},
            'heating': {'r': 0.3, 'g': 0.18, 'b': 0.0, 'w': 0.0},
            'homing': {'r': 0.0, 'g': 0.6, 'b': 0.2, 'w': 0.0},
            'leveling': {'r': 0.5, 'g': 0.1, 'b': 0.4, 'w': 0.0},
            'meshing': {'r': 0.2, 'g': 1.0, 'b': 0.0, 'w': 0.0},
            'off': {'r': 0.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'printing': {'r': 0.0, 'g': 0.0, 'b': 1.0, 'w': 1.0},
            'standby': {'r': 0.1, 'g': 0.0, 'b': 0.1, 'w': 0.1},
        },
        'nozzle': { # Colors for nozzle states
            'heating': {'r': 0.8, 'g': 0.35, 'b': 0.0, 'w':0.0},
            'off': {'r': 0.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'on': {'r': 1.0, 'g': 1.0, 'b': 1.0, 'w':1.0},
            'standby': {'r': 0.1, 'g': 0.0, 'b': 0.1, 'w':0.0},
        },
        'thermal': {
            'hot': {'r': 1.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'cold': {'r': 0.3, 'g': 0.0, 'b': 0.3, 'w': 0.0}
        }
    }
variable_logo_led_name:         "sb_leds" 
variable_logo_idx:              "1" 
variable_nozzle_led_name:       "sb_leds"
variable_nozzle_idx:            "2,3"
gcode:
    # This section is required.  Do Not Delete.

[gcode_macro _set_sb_leds]
gcode:
    {% set red = params.RED|default(0)|float %}
    {% set green = params.GREEN|default(0)|float %}
    {% set blue = params.BLUE|default(0)|float %}
    {% set white = params.WHITE|default(0)|float %}
    {% set led = params.LED|string %}
    {% set idx = (params.IDX|string).split(',') %}
    {% set transmit_last = params.TRANSMIT|default(1) %}

    {% for led_index in idx %}
        {% set transmit=transmit_last if loop.last else 0 %}
        set_led led={led} red={red} green={green} blue={blue} white={white} index={led_index} transmit={transmit}
    {% endfor %}

[gcode_macro _set_sb_leds_by_name]
gcode:
    {% set leds_name = params.LEDS %}
    {% set color_name = params.COLOR %}
    {% set color = printer["gcode_macro _sb_vars"].colors[leds_name][color_name] %}
    {% set led = printer["gcode_macro _sb_vars"][leds_name + "_led_name"] %}
    {% set idx = printer["gcode_macro _sb_vars"][leds_name + "_idx"] %}
    {% set transmit = params.TRANSMIT|default(1) %}

    _set_sb_leds led={led} red={color.r} green={color.g} blue={color.b} white={color.w} idx="{idx}" transmit={transmit}

[gcode_macro _set_logo_leds]
gcode:
    {% set red = params.RED|default(0)|float %}
    {% set green = params.GREEN|default(0)|float %}
    {% set blue = params.BLUE|default(0)|float %}
    {% set white = params.WHITE|default(0)|float %}
    {% set led = printer["gcode_macro _sb_vars"].logo_led_name %}
    {% set idx = printer["gcode_macro _sb_vars"].logo_idx %}
    {% set transmit=params.TRANSMIT|default(1) %}

    _set_sb_leds led={led} red={red} green={green} blue={blue} white={white} idx="{idx}" transmit={transmit}

[gcode_macro _set_nozzle_leds]
gcode:
    {% set red = params.RED|default(0)|float %}
    {% set green = params.GREEN|default(0)|float %}
    {% set blue = params.BLUE|default(0)|float %}
    {% set white = params.WHITE|default(0)|float %}
    {% set led = printer["gcode_macro _sb_vars"].nozzle_led_name %}
    {% set idx = printer["gcode_macro _sb_vars"].nozzle_idx %}
    {% set transmit=params.TRANSMIT|default(1) %}

    _set_sb_leds led={led} red={red} green={green} blue={blue} white={white} idx="{idx}" transmit={transmit}

[gcode_macro set_logo_leds_off]
gcode:
    {% set transmit=params.TRANSMIT|default(1) %}
    _set_logo_leds red=0 blue=0 green=0 white=0 transmit={transmit}

[gcode_macro set_nozzle_leds_on]
gcode:
    {% set transmit=params.TRANSMIT|default(1) %}
    _set_sb_leds_by_name leds="nozzle" color="on" transmit={transmit}

[gcode_macro set_nozzle_leds_off]
gcode:
    {% set transmit=params.TRANSMIT|default(1) %}
    _set_sb_leds_by_name leds="nozzle" color="off" transmit={transmit}

[gcode_macro status_off]
gcode:
    set_logo_leds_off transmit=0
    set_nozzle_leds_off

[gcode_macro status_ready]
gcode:
    _set_sb_leds_by_name leds="logo" color="standby" transmit=0
    _set_sb_leds_by_name leds="nozzle" color="standby" transmit=1

[gcode_macro status_busy]
gcode:
    _set_sb_leds_by_name leds="logo" color="busy" transmit=0
    set_nozzle_leds_on

[gcode_macro status_heating]
gcode:
    _set_sb_leds_by_name leds="logo" color="heating" transmit=0
    _set_sb_leds_by_name leds="nozzle" color="heating" transmit=1

[gcode_macro status_leveling]
gcode:
    _set_sb_leds_by_name leds="logo" color="leveling" transmit=0
    set_nozzle_leds_on

[gcode_macro status_homing]
gcode:
    _set_sb_leds_by_name leds="logo" color="homing" transmit=0
    set_nozzle_leds_on

[gcode_macro status_cleaning]
gcode:
    _set_sb_leds_by_name leds="logo" color="cleaning" transmit=0
    set_nozzle_leds_on

[gcode_macro status_meshing]
gcode:
    _set_sb_leds_by_name leds="logo" color="meshing" transmit=0
    set_nozzle_leds_on

[gcode_macro status_calibrating_z]
gcode:
    _set_sb_leds_by_name leds="logo" color="calibrating_z" transmit=0
    set_nozzle_leds_on

[gcode_macro status_printing]
gcode:
    _set_sb_leds_by_name leds="logo" color="printing" transmit=0
    set_nozzle_leds_on
`

Pitufo123, 2024-06-30:
Anyone figure this out?  I've tried connecting the signal line to the 4w fan pwm as well but no matter what I've tried can't get mine to work

mateusdemboski, 2024-08-01:
Same problem...

@bigtreetech could you help here?

Pitufo123, 2024-08-01:
In my case the problem ended up being faulty neopixels in my siboor kit.  Ordered a new string and they worked just fine

mateusdemboski, 2024-08-01:
I found a workaround on Reddit that resumes in use fan PWM pin instead of RGB pin. It worked for me.

bigtreetech, 2024-08-02:
The RGB pin is protected via a 100ohm resistor at the output. This ensures that the pin is not damaged if it is shorted to something that it should not be. If you have problems with a particular RGB strip then it could be that the strip has an input resistance that is too low on the signal line (it should be incredibly high so that there is no volt drop on a proper RGB strip). You could try to remove and short this resistor out if it is affecting you but it is important to know that you lose protection on that pin so don't short it. The resistor is R55

[image]

closed 2024-08-02 by bigtreetech

skyguy94, 2024-09-01:
@bigtreetech I can't find a document that tells you which resistor is R55, nor is there anything on the screen print of the SB0000 that indicates which is R55. The SMD components/board layout are tight enough that I'm having difficulty measuring the impedance of the resistors to determine which is the 10k and which is the 100 ohm.

The only references I can find of the SB0000 are in the build guide and the 3D PDF model. Can you post a picture with R55 labelled?

bigtreetech, 2024-09-03:
> The only references I can find of the SB0000 are in the build guide and the 3D PDF model. Can you post a picture with R55 labelled?

Here it is. Please let me know if it works out for you. It will be interesting if some RGB LED variants have been manufactured with very low input impedance.

[image]

skyguy94, 2024-09-06:
Ok, I removed the resistor and soldered a small piece of wire in its place and the neopixels did not turn on. Its fair to say that I could have damaged the pads, traces, or components and introduced a different problem, but I don't have a way of probing something this tightly packed to test. However, I did switch to my spare EBB2209 and EBB SB0000 those worked without an issue.

Pitufo123, 2024-09-29:
I'm having the exact same issue but this time it's not my neopixels I've got a brand new board.  Fair to assume it's faulty?  I read somewhere there's a batch of faulty boards on Amazon.

jchodakowski, 2024-09-29:
@Pitufo123 where did you source your new board and how much of the set did you replace?

Pitufo123, 2024-09-29:
I bought the board(s) off amazon as well as the neopixel chains. I tried only replacing the sb2208 board and theb also swapped out the 0000 daughter board.  Everything else works except for the neopixels.  And I've checked for broken wires on the neopixel chain.

tsoos99dev, 2025-01-08:
I have a similar issue. First, I burned out the Schmitt-trigger for the 4 wire fan input, and then I bent the main power input while testing the bed corner locations. I finished my first print at least, so I'll look into what's going on with the LEDs. I can't change their colour at all. I'll check the power and data pins tomorrow, and also verify that the LEDs are fine separately.

dutchronnie, 2025-01-12:
It was a faulty chip on the SB0000, i have replaced the chip for a new one,
and everything was working again

Dummykid, 2025-01-14:
well,
1st try: I bought brand new Bigtreetech SB LED light, and it didn't work.
2nd try: I bought brand new SB0000, and it didn't work.
3rd try: I removed the "yellow-signal" pin from the LED dupoint connector, and connected it to the fan PWM port and it worked. 

I think BTT need to have quality control on its product. I have read all the posts here and everyone got different issue on this LED light.

sdylewski, 2025-02-02:
> It was a faulty chip on the SB0000, i have replaced the chip for a new one,
> and everything was working again
> 
> Op wo 8 jan 2025 om 23:28 schreef Soós Tamás ***@***.***>:
> […](#)

Which chip was this?  the SN74LVC1G17?

sdylewski, 2025-02-04:
I have the same problem, and tried shorting the resistor R55, but still doesn't work.  When I first connected my LED board, my lights worked, but stopped during the first print. Is it possible I have a faulty SN74LVC1G17?

dutchronnie, 2025-02-04:
Yes it was that chip,
[image: image.png]
Bought this one to replace: https://nl.aliexpress.com/i/32843534645.html

ptamarin, 2025-05-03:
Also having this issue with GPIO16 pin, just testing my neopixel leds (with raspberry pi 3b via its GPIO) and can rule them out. Leds and BTT SB2209 RP2040 from a formbot 2.4 kit