bigtreetech/EBB #80: EBB42 V1.2 part cooling fan not working

SooscLick, 2023-10-06:
I've just successfully installed the board and everything including the hotend fan works fine except the part cooling fan. 

It's not moving at values from 0 to 255 and showing 0V output at 255 (full throttle).

Reversing the pins in the .cfg (PA0 for hotend fan and PA1 for part cooling) makes part cooling fans spin at 100% while hotend fan now inoperable so there's no electrical problem.

Changing pwm to hardware, cycle and kickstarting doesn't change anything.

SooscLick, 2023-10-08:
Found out it's lacking hardware pwm on EBB42 required for sunon fans

adjustinthings, 2023-12-16:
I figured it out. You still have your old [fan] object in your printer.cfg file. Remove that. Good to go.