bigtreetech/EBB #86: filament sensor EBB SB2209 CAN(RP2040)

Krlesul, 2023-11-23:
Hi, can you help me please?
I have no idea to which connector on the sb2209 board I can connect filament sesor which I have in the tool head Stealthburner.

Thank you so much

cruzer619, 2023-12-05:
try using gpio6 with jumper IND on and NPN jumper on.

cruzer619, 2023-12-05:
[image]

Krlesul, 2024-02-10:
> Did it work?

Yes, Thank you

*S pozdravem a přáním krásného dne *
*Best regards and wish a beautiful day*

*František Kršňák*
* Czech Republic*

cruzer619, 2024-02-10:
thats good to know it worked with the workaround.

Hulign, 2024-02-28:
This is helpful info. Is there another spot that works? I have one before gears and after in my galileo 2 setup.

cruzer619, 2024-02-28:
> This is helpful info. Is there another spot that works? I have one before gears and after in my galileo 2 setup.

@Hulign nope thats the only "extra" port available to my knowledge.

hartmanrik, 2024-03-05:
> > This is helpful info. Is there another spot that works? I have one before gears and after in my galileo 2 setup.
> 
> @Hulign nope thats the only "extra" port available to my knowledge.

I intend to use 2 pins on the probe connector (I use a cartographer3d scanner via CAN as bed probe). That way I think it must be possible…

iwm911, 2024-04-09:
i think its possible to use the 2 extra fans GPIOs (if not in use)

dearnheart, 2024-04-11:
Going to be installing a ERCF soon. Here is how I connected the two runout sensors.
[image]

cruzer619, 2024-04-11:
@dearnheart nice... didn't realize SB2209 has 2 endstop ports.. (didn't really look at the schematics of the SB2209 thought they would be similar to SB2209-RP2040 version.  Just didn't connectors and chip)..  as I'm thinking of getting an ERCF v2 as well..  Now I have to think do I need to buy that version..

dearnheart, 2024-04-11:
> @dearnheart nice... didn't realize SB2209 has 2 endstop ports.. (didn't really look at the schematics of the SB2209 thought they would be similar to SB2209-RP2040 version. Just different connectors and chip).. as I'm thinking of getting an ERCF v2 as well.. Now I have to think do I need to buy that version..

Give BTT horrible naming for these boards I can easily see the confusion.

fkahhaleh, 2024-04-16:
> Going to be installing a ERCF soon. Here is how I connected **the two runout sensors**. [...]

Hi dearnheart, by the 'two runout sensors' are you referring to the Toolhead and Entry Sensor?
Thanks!

dearnheart, 2024-04-16:
> Hi dearnheart, by the 'two runout sensors' are you referring to the Toolhead and Entry Sensor? Thanks!

Correct, I was planning on using this mod.

https://github.com/Enraged-Rabbit-Community/ERCF_v2/tree/master/Recommended_Options/ERF_Filament_Cutter/Stls/1_Toolhead_And_Entry_Sensors

fkahhaleh, 2024-04-16:
> Correct, I was planning on using this mod.
>
> https://github.com/Enraged-Rabbit-Community/ERCF_v2/tree/master/Recommended_Options/ERF_Filament_Cutter/Stls/1_Toolhead_And_Entry_Sensors

Great. I am at the road fork of deciding which route to go.
Did you actually make the connections/test it working? or just planning phase!
Thank you

dearnheart, 2024-04-16:
> Great. I am at the road fork of deciding which route to go. Did you actually make the connections/test it working? or just planning phase! Thank you

Connection was made and fully functional.
[image]
[filament_switch_sensor filament_sensor2]
switch_pin: EBBCan: PB8
pause_on_runout: True
insert_gcode:
    M117 Insert Detected
runout_gcode:
    M117 Runout Detected
    #LCDRGB R=1 G=0 B=0  # Turn LCD red
    #BEEP I=12

[filament_switch_sensor filament_sensor]
switch_pin: EBBCan: PC13
pause_on_runout: True
insert_gcode:
    M117 Insert Detected
runout_gcode:
    M117 Runout Detected
    #LCDRGB R=1 G=0 B=0  # Turn LCD red
    #BEEP I=12

fkahhaleh, 2024-04-16:
Wow this is really good news!
Thank you for sharing your setup.

mrellifant, 2024-06-04:
> I intend to use 2 pins on the probe connector (I use a cartographer3d scanner via CAN as bed probe). That way I think it must be possible…

Did this work? I’m looking to do the same.. cartographer via can on sb2209 (sb2040) 2 sensors on the probe connector. Thanks

fkahhaleh, 2024-06-05:
> Did this work? I’m looking to do the same.. cartographer via can on sb2209 (sb2040) 2 sensors on the probe connector. Thanks

Yes, all the details shared by @dearnheart are spot on and worked for me.

Pitufo123, 2024-08-04:
O no I was gonna do this too but sb2209 and sb2209 rp2040 are 2 different boards??  Is it not possible to put 2 sensors on sb2209 rp2040?

dearnheart, 2024-08-04:
> O no I was gonna do this too but sb2209 and sb2209 rp2040 are 2 different boards?? Is it not possible to put 2 sensors on sb2209 rp2040?

Wish I could say, I can only speak to the board I have.

iwm911, 2024-08-05:
> O no I was gonna do this too but sb2209 and sb2209 rp2040 are 2 different boards?? Is it not possible to put 2 sensors on sb2209 rp2040?

i use BL Touch for bed leveling and i am able to use 2 sensors on the 2209 rpi2040.
i used the spare GPIO (i think its GPIO6) and the extra fan on the cover plate using the PWM pin.

Pitufo123, 2024-08-05:
I've got gpio6 and gpio21 currently (on sb2209 rp2040 board). One of them is always showing detected and the other shows detected when I insert filament but then reverts to empty with filament still inserted.  I'm not certain if one of the switches is stuck or not, but what would cause the detected state to revert with switch still triggered?

dearnheart, 2024-08-05:
> I've got gpio6 and gpio21 currently (on sb2209 rp2040 board). One of them is always showing detected and the other shows detected when I insert filament but then reverts to empty with filament still [...]

add a ! to the pin to reverse it
"Pin names may be preceded by ! to indicate that a reverse polarity should be used (eg, trigger on low instead of high)."

Pitufo123, 2024-08-05:
Ya but that would just invert the state right?  What's happening on one of my sensors is that I insert the filament, the state changes, and then without moving the filament, the state reverts to the previous state

dearnheart, 2024-08-05:
> Ya but that would just invert the state right? What's happening on one of my sensors is that I insert the filament, the state changes, and then without moving the filament, the state reverts to the previous state

Oh, sounds like maybe a bad sensor or something, I am at a loss on that one.

Pitufo123, 2024-08-05:
Ya I'm really confused...  I've got a sticky top sensor (ball bearing not working properly) but when I take it apart and check with a multimeter both connections work.  I have them wired NC.  The bottom sensor on gpio6 doesn't change state in klipper. (The sensor that reliably works with the multimeter)

dearnheart, 2024-08-05:
Could be the jumpers on the board, could be the switch is loose to the point it triggers when the filament enters but the pressure pushes it back, could be the wiring. To many variables

kabelrolle, 2024-08-08:
> I've got gpio6 and gpio21 currently (on sb2209 rp2040 board). One of them is always showing detected and the other shows detected when I insert filament but then reverts to empty with filament still [...]

I have the same situation... already got a new sensor but the problem still exists... i wonder if there is a jumper missing?
I will try the other GPIOs mentioned above, maybe they will work...

iwm911, 2024-08-08:
> I have the same situation... already got a new sensor but the problem still exists... i wonder if there is a jumper missing? I will try the other GPIOs mentioned above, maybe they will work...

there is the jumper that sets the GPIO to fan or ind. it should be on ind
[image]

Pitufo123, 2024-08-08:
I have mine on ind

andrewmherren, 2024-11-10:
Just did this on my sb2209 (rp2040) with tap, toolhead sensor, and entry sensor so I figured id add the full diagram of the wiring I did that worked since this issue is a top google result and has a really good collection of info.

I did a breakout cable to turn the probe connector into one cable for tap and one for a sensor. Obviously you could swap the x, toolhead, and entry sensors anyway you want, this is just how I wired them. For all 3 sensors I only used ground and the GPIO, no +5v.

[image]

mrellifant, 2024-11-11:
Awesome.

image.png (view on web)

johyphenel, 2024-12-18:
@andrewmherren did you have to install a physical pullup resistor between gpio21 and a 5v pin? That's the only way I could get the gpio21 pin to work with a switch or filament sensor on my board. 

Ended up using gpio6 for filament sensor (with IND + NPN jumpers on), since I didn't want to bother with a permanent resistor (at least right now).

andrewmherren, 2024-12-18:
@johyphenel your timing is impeccable... I set this all up and tested it in preparation for my ERCF which I finally completed this weekend. After about an hour of use the sensor on gpio21 stopped working. I guess it was briefly floating high enough when untriggered to work (for awhile)? Anyway, you are correct, it seems that if you want to use gpio21 you need a physical pullup resister. I didn't want to stop and mess with that in the middle of my ERCF work so I just moved to gpio27 on TH0 which has been working fine. Thanks bringing this back up, i'll update my picture/comment above so as not to trick anyone else.

zeprodk, 2024-12-28:
@andrewmherren 
Hey i have everything for the erf with 2 switches and ercf. The formbot kit i have uses tap and endstop but doesnt use the th0 for the thermistor so i would like to know if this setup is working well for you?

andrewmherren, 2024-12-28:
@zeprodk Yes, I haven't bothered to switch back so far, so i'm still using th0 and haven't had any problems since making the change last week. I haven't done a ton of printing this past week but I have done some and have been working on tuning ercf all day and haven't had any issues with the toolhead/extruder sensors.

zeprodk, 2024-12-28:
@andrewmherren Thank you i will try this when i start the build!

popelavojtech, 2024-12-29:
I was just going through it today too. 

My setup:
SBB2209 RP2040, 2 toolhead filament sensors for ercf, X endstop on the octopus, CNC TAP on ebb (gpio22).

NPN jumper in place, IND also, pullup resistor 4k7 wired physically between gpio21 and +5V.
I´m using gpio21 for toolhead_runout, gpio6 for hot_end_entry (gnd on gnd, second pins on gpio).
It works like a charm!

jace, 2025-01-15:
I'm setting up ERCF v2 with Galileo 2 extruder ([with this mod for Omron D2F-L3-D3 switches without bearings](https://github.com/juliusjj25/G2E-Filametrix-Lever-Switch-Mod)), Rapido 2 hotend, and an SB2209 RP2040 CAN toolhead board. The comments here have been most helpful!

I need two sensors (toolhead and entry). My port allocation:

1. Endstop (gpio24): taken by X-axis endstop
2. TH0 (gpio27): taken by Rapido 2's PT1000 thermistor
3. Probe (gpio21-22): 21 is unoccupied, but needs a breakout since the other pin (gpio22) is taken by Voron Tap's Z-axis endstop, plus a pull-up resistor as @andrewmherren noted (I have no idea how to do that… yet)
4. IND/FAN (gpio6): unoccupied
5. HE0 (gpio7): taken by hotend heater

I guess my options are gpio6 and one of:

1. [ ] gpio7, and move hotend's thermistor to the next port (gpio8-11), or
2. [ ] gpio21, with breakout and pull-up resistor

(I'll update as I figure this out.)

jace, 2025-01-26:
I've had recurring issues with Voron Tap probe misreading as triggered since I made this change. It's not consistent and most often shows up after a long print when I try to home and start the next print. This sounds like the pullup-needed problem @Pitufo123 and @andrewmherren reported with `gpio21`, but my Voron Tap probe is still in `gpio22` as before. Endstop readings are slow enough that I'm able to capture screenshots showing Endstop Z and Probe (virtual endstop z) in different states (they'll sync up after about a second, but too slow for homing):

[image] [image]

The only thing I've changed is adding the jumpers for NPN and IND. What does this NPN jumper do? Is it causing `gpio22` misreadings? I don't understand this explanation in the manual:

> Supports both NPN and PNP types. For NPN, short the two pins at "NPN" using a jumper; for PNP, remove the jumper.
Note: This interface can also be used for 24V CNC fans. By shorting the left two-pin connector with a jumper (FAN), it can be
used for a 24V CNC fan. By shorting the right two-pin connector with a jumper (IND), it can be used for a 24V proximity switch.

**Possibly related:** After moving the hotend's thermistor to the 31865 I2C port, I've had very occasional misreadings of the thermistor when idle. It'll jump to >10k°C and then rapidly drop towards normal. This has not caused Klipper to shutdown for exceeding max_temperature, oddly. I'm assuming I have a loose contact, either in the crimping (fabric-wrapped wire is damned hard to crimp!) or from putting a 2-pin JST PH plug in a 4-pin socket.

Since I've struggled for months with CAN bus "timer too close" errors that were only resolved by reducing the LED effects plugin's framerate to < 12 fps, I'm now wondering if the thermistor's mis-readings are causing MCU overloads and affecting probe readings. Here's a section of klippy.log from when I was testing endstop triggers:

```
probe: open
Stats 97.3: gcodein=0  mcu: mcu_awake=0.003 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=27835 bytes_read=44328 bytes_retransmit=9 bytes_invalid=0 send_seq=1785 receive_seq=1785 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000140 toolhead: mcu_awake=0.017 mcu_task_avg=0.000022 mcu_task_stddev=0.000047 bytes_write=19207 bytes_read=29833 bytes_retransmit=0 bytes_invalid=0 send_seq=1623 receive_seq=1623 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=12000166 adj=12000193  heater_bed: target=0 temp=60.1 pwm=0.000 Raspberry_Pi: temp=48.2 mcu: temp=43.2 toolhead_mcu: temp=56.9 Chamber: temp=33.3 sysload=1.02 cputime=11.549 memavail=430448 print_time=103.283 buffer_time=0.000 print_stall=0 extruder: target=0 temp=49.7 pwm=0.000
Stats 98.3: gcodein=0  mcu: mcu_awake=0.003 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=28009 bytes_read=44796 bytes_retransmit=9 bytes_invalid=0 send_seq=1798 receive_seq=1798 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000151 toolhead: mcu_awake=0.017 mcu_task_avg=0.000022 mcu_task_stddev=0.000047 bytes_write=19508 bytes_read=30164 bytes_retransmit=0 bytes_invalid=0 send_seq=1647 receive_seq=1647 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=12000165 adj=12000196  heater_bed: target=0 temp=60.0 pwm=0.000 Raspberry_Pi: temp=48.2 mcu: temp=43.1 toolhead_mcu: temp=56.9 Chamber: temp=33.3 sysload=1.02 cputime=11.593 memavail=430448 print_time=103.283 buffer_time=0.000 print_stall=0 extruder: target=0 temp=49.6 pwm=0.000
Stats 99.3: gcodein=0  mcu: mcu_awake=0.003 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=28190 bytes_read=45239 bytes_retransmit=9 bytes_invalid=0 send_seq=1812 receive_seq=1812 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000149 toolhead: mcu_awake=0.015 mcu_task_avg=0.000022 mcu_task_stddev=0.000046 bytes_write=19828 bytes_read=30538 bytes_retransmit=0 bytes_invalid=0 send_seq=1674 receive_seq=1674 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=2 freq=12000166 adj=12000172  heater_bed: target=0 temp=60.0 pwm=0.000 Raspberry_Pi: temp=48.2 mcu: temp=43.2 toolhead_mcu: temp=56.8 Chamber: temp=33.2 sysload=1.02 cputime=11.640 memavail=430448 print_time=105.543 buffer_time=0.211 print_stall=0 extruder: target=0 temp=49.7 pwm=0.000
probe: TRIGGERED
Stats 100.3: gcodein=0  mcu: mcu_awake=0.003 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=28364 bytes_read=45678 bytes_retransmit=9 bytes_invalid=0 send_seq=1825 receive_seq=1825 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000139 toolhead: mcu_awake=0.015 mcu_task_avg=0.000022 mcu_task_stddev=0.000046 bytes_write=20136 bytes_read=30890 bytes_retransmit=0 bytes_invalid=0 send_seq=1699 receive_seq=1699 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=12000166 adj=12000178  heater_bed: target=0 temp=60.0 pwm=0.000 Raspberry_Pi: temp=48.7 mcu: temp=43.1 toolhead_mcu: temp=56.7 Chamber: temp=33.3 sysload=0.94 cputime=11.696 memavail=430448 print_time=105.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=49.6 pwm=0.000
Stats 101.3: gcodein=0  mcu: mcu_awake=0.003 mcu_task_avg=0.000006 mcu_task_stddev=0.000004 bytes_write=28538 bytes_read=46146 bytes_retransmit=9 bytes_invalid=0 send_seq=1838 receive_seq=1838 retransmit_seq=2 srtt=0.000 rttvar=0 [... cut ...]
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
    RX: bytes  packets  errors  dropped missed  mcast
    18557689   2968715  0       0       0       0
    TX: bytes  packets  errors  dropped carrier collsns
    38875008   5409701  0       0       0       0
```

So, three suspects to find the culprit from:

1. NPN jumper is causing `gpio22` to misread
2. Loose contact in the probe plug
3. Loose contact in the thermistor is causing MCU overloads

cruzer619, 2025-01-31:
@jace just noticing your pinning for voron tap.. ( I have chaotic tap ) is wired differently then mine.  my ground wire is at pin 1 on connector P9.  and I'm using pin 5 (gpio21) and 4 for tool head entry sensor.  And also noticed your missing the jumper on IND for gpio6. I would try those and see.

jjkelly1992, 2025-03-07:
> I intend to use 2 pins on the probe connector (I use a cartographer3d scanner via CAN as bed probe). That way I think it must be possible…

Where did you connect your Cartographer?  Having to connect up front on the SB0000 board is really not an elegant solution to me.  I was hoping that because the SB2209 RP2040 is software CAN that I might be able to assign the GPIO pins in the "Probe" connector to be CAN-H and CAN-L for Cartographer3D.    Can't seem to find any information if that it possible.

TurboTronix, 2025-04-25:
@jace which board did you buy? SB2209 RP2040 or CAN?

jace, 2025-04-26:
> @jace which board did you buy? SB2209 RP2040 or CAN?

SB2209 RP2040, which I replaced with SB2209 USB, which came with a defective cable and didn't work either. I fixed that cable (cut it open and re-soldered the wires) and it _mostly_ worked. Then I got another SB2209 USB and took the cable from that and reliability was better, and then BIQU finally shipped me a replacement cable. Now I have enough spares to build a second printer.

The good news is `gpio21` on the SB2209 USB does not need an external pull-up, so I have both filament sensors working properly.

oleksii-suprun, 2025-05-19:
I think it makes sense to link this issue to another directly related to the BLTouch connector and the `gpio21`. People also have issues with it and to make it work, there also a recommendation to add an extra physical pullup resistor for the `gpio21`: https://github.com/bigtreetech/EBB/issues/88

brandanbooth, 2025-08-02:
> Going to be installing a ERCF soon. Here is how I connected the two runout sensors. [image]

So I just got done building my Voron 2.4. What was your Klipper config for the TAP using the probe port that is meant for the bttouch?

MamedovArtur, 2025-10-31:
Just switch servo pin and sensor pin, it works for me