bigtreetech/EBB #88: EBB SB2209 RP2040 : Bltouch always open

cruzer619, 2023-11-29:
Having issue with Bltouch on newly installed EBB SB2209 RP2040 on manta m8p canbus.  I'm able to manually pin up and pin down.  but wont sense pin up doing touch mode.  everything else works on the ebb, thermistor, heater cart, fans, limit switch and extruder motor.  Please help.  Thanks.

cruzer619, 2023-12-04:
ok if anyone searches this.  This is what I had to do get bltouch working.  As it seems that gpio21 has issues as with others in the Tap area are complaining as well.  Since bltouch uses gpio22 for control can't just switch to like the tap guys.  So I had to move things around.  I had to gpio21 wire to gpio24.  Then for xstop I used gpio6 but you have to jumper on IND and jumper on NPN.

Ixian, 2024-01-19:
I don't know if this can be re-opened but it really shouldn't be closed - the problem still exists, and while it is possible to tap another gpio port that is A) a hack and B) takes away use of that port. Right now I have to use two ports for my CR Touch which means I also had to choose between an end stop and a filament sensor. If this worked like it should I wouldn't have this problem.

The board is shipping broken and BTT needs to address it. From the looks of it, gpio21 either doesn't work with a pullup set, or it is too weak, meaning the state will always show 1 (closed) because of the float. I can try playing around with external pullup resistors, though that isn't the long term solution either.

cruzer619, 2024-01-19:
@Ixian think your right.. will leave it open as there needs to be a fix.

Gardosen, 2024-02-24:
For People like me who uses the Endstop for the X Endstop, it is not even a useable workaround.... 

this is really fucked up to be honest.....

cruzer619, 2024-02-27:
> For People like me who uses the Endstop for the X Endstop, it is not even a useable workaround....
> 
> this is really fucked up to be honest.....

actually this workaround uses the proximity port/3rd fan port.  If you need that port then this workaround doesn't work for you.  As we are using that port for X endstop.  Still this sucks as we are not uses the indented port for what its suppose to be used for.  But i'm glad I found a workaround or I would have a useless device.
[image]

thatguywhodoesthethings, 2024-04-14:
> I have put the crtouch into GPIO6 and it works so at least i got a working printer again, but sacrificing the filament sensor.
> 
> Has anyone tried putting a 10k resistor between vcc and ground like diagram B here? - https://reprap.org/wiki/File:Reprap_endstops_schem_mech.png
> 
> I might try this with a normal endstop, and if it works, at least we can get a working pin

Have you gotten a chance to try yet?

kenny000666, 2024-04-14:
No, didn’t work. Wired in a 10k resistor and didn’t get the desired results. Swapped it out for the non-rp2040 SB2209 in the end

cruzer619, 2024-04-14:
@thatguywhodoesthethings the other you can do is ditch the bltouch.  ( what i'm thinking of doing, want to add ERCF v2 to my printer ) and get klicky or tap type probe.  Won't need a sensing pin.  And then you have enough ports for filament sensor.  So you would use bltouch port for your klicky/tap gpio22 only.  then xport for x endstop.  Then proximity port for Filament sensor.

kenny000666, 2024-04-14:
Yes, I considered that also. Similar to the Tap workaround and use GPIO22 as the z-sensor. I also have Ercfv2 on my printer, and it’s better if you have 2 sensors for the extruder and toolhead. The non-Rp2040 version has enough working pins (and then some) out of the box to solve this problem, so I went that direction.

cruzer619, 2024-04-14:
hrmmm good point.  but can work with only 1 sensor on the toolhead for ercf..  oh and there is no issue with the bltouch port on the non-rp2040 version... it all works right??

kenny000666, 2024-04-14:
I figured whilst I had the ports and microswitches, why not use the 2 sensor version of clockwork2.

Yes, the non-rp2040 version works with BLtouch with no issues. Main downside to this version is the connectors are way smaller and a PITA to crimp. Also the temperature sensor port is blocked by the clockwork 2 case so you gotta remove the the sb2209 to plugin the temp sensor. Makes hot swapping or maintenance difficult. Otherwise, functionality wise, it all works

drowfx, 2024-05-20:
Looking at the schematic it is clear why this cannot work: 

- the rp2040 gpio21 has the internal pullup, which is what is activated when the pin is declared as ^EBBCan:gpio21 in klipper
- between the gpio21 and the external connector (P9, pin 5; or the leftmost pin of the connector labeled "Probe") sits a 74LVC2G17DW buffer (U13) in the schematic. this is a unidirectional schmitt-trigger buffer which is here used to make the input 5V-safe, which the rp2040 is not
- but this means that any internal pullup set via klipper on the rp2040 is totally useless - the input is always actively driven by the buffer chip. To work with an external switch (filament sensor, endstop, ...) that simply closes against GND in one state and is open in the other, a pullup resistor is needed on the input side of the buffer.

One workaround is to solder a ~10 kOhm resistor between pins 2 and 5 of the P9 connector. This adds the necessary pullup and should be moderately easy to solder on the back of the board.

A real fix would have to be done by BTT (add a pullup and a jumper to enable/disable it, at the input side of U13). Or at least this limitation should be clearly documented.

julien-f-leroy, 2024-06-02:
@drowfx I tried your workaround to solder a 10kOhm resistor between the pin 2 and 5 of the P9 and it worked perfectly.

kgadberry, 2024-06-04:
@drowfx So if I understand correctly, pin 5 (gpio21) _is_ 5v-tolerant?

Wouldn't it possible then to just configure the probe to output 5v to the signal pin instead of operating in open drain mode? (Assuming the probe has that capability, which I believe the BLTouch 3.0 does.)

That said, connecting a 10k resistor between pins 2 and 5 does work, and I appreciate the insight.

drowfx, 2024-06-04:
> @drowfx So if I understand correctly, pin 5 (gpio21) _is_ 5v-tolerant?

The gpio21 of the rp2040 controller itself is _not_ 5V tolerant. The pin 5 of the external connector is not directly connected to the rp2040, but there is a chip inbetween that translates the external 5V input to 3.3V for the rp2040. That makes the external input (pin 5 of P9) 5V tolerant.

> Wouldn't it possible then to just configure the probe to output 5v to the signal pin instead of operating in open drain mode? (Assuming the probe has that capability, which I believe the BLTouch 3.0 does.)

Yes, if your probe supports push-pull mode that would work too, without the external pull up resistor. My suggestion with the resistor was for cases where the probe/sensor only works in open drain mode, or a simple switch is used which shorts the pin against GND when closed (e.g. filament sensor).

> That said, connecting a 10k resistor between pins 2 and 5 does work, and I appreciate the insight.

Thanks.

cruzer619, 2024-06-04:
going to try the 10k workaround this weekend.. will be nice to have more ports available for when I get ercf v2..

IggyTheWolf, 2024-06-23:
I'm running into the same issue. can someone point me in the right direction as which are pin 5 and 9?

kgadberry, 2024-06-23:
> I'm running into the same issue. can someone point me in the right direction as which are pin 5 and 9?

You're looking for pins 2 and 5 of the connector that's labeled P9 in the schematic, which is the "probe" connector on the board.

See [this diagram](https://github.com/bigtreetech/EBB/blob/69b5881b9d55ca2d5bf0a14f423dce30feaceaec/EBB%20SB2209%20CAN%20(RP2040)/Hardware/EBB%20SB2209%20CAN%20V1.0%EF%BC%88RP2040%EF%BC%89-Pin.pdf).

You want to connect a pull-up resistor between +5v and (pin 5 which connects to) GPIO21. This brings the pin to a default state of logic high and the probe completes the path to ground when triggered which brings the pin to logic low. Without the resistor the pin can only be logic low unless the probe itself applies +5v, as the BLTouch/CRTouch can be configured to do.

IggyTheWolf, 2024-06-23:
@kgadberry Thanks! I'll add a 10K between the two. I'm surprised this hasn't been fixed on the board or microprobe.

IggyTheWolf, 2024-06-24:
I added a 10K resistor between pin 2 and 5 (5v and GPIO21) and either the probe is always triggered when deployed or always open.

I'm going to try and use GPIO6 and see if that will work in place of GPIO21

cruzer619, 2024-06-24:
> I added a 10K resistor between pin 2 and 5 (5v and GPIO21) and either the probe is always triggered when deployed or always open.
> 
> I'm going to try and use GPIO6 and see if that will work in place of GPIO21
> 
> EDIT: Same result with GPIO6

if you hookup the bltouch directly to the main board.. do you get the same results... I thought I was having an issue with the sb2209 turned out to be the bltouch.. ( but mine was a clone, got OG and worked as expected directly from mainboard.  and then worked from sb2209 )

IggyTheWolf, 2024-06-24:
@cruzer619 I'm using the MicroProbe, but I can check to see if it works plugged in directly to the board. Right now the Microprobe deploys and stows just fine, it just either registers fully open or closed, even with a 10K resistor on GPIO21 and connected to GPIO6

looxonline, 2024-06-24:
> @cruzer619 I'm using the MicroProbe, but I can check to see if it works plugged in directly to the board. Right now the Microprobe deploys and stows just fine, it just either registers fully open or closed, even with a 10K resistor on GPIO21 and connected to GPIO6

The microprobe will always be triggered when stowed. It is only not triggered when deployed.

IggyTheWolf, 2024-06-24:
@looxonline right, but mine is either always triggered even when deployed, or always open even when stowed.

Here's my config:

`[gcode_macro PROBE_DOWN]
gcode:
    SET_PIN PIN=probe_enable VALUE=1

[gcode_macro PROBE_UP]
gcode:
    SET_PIN PIN=probe_enable VALUE=0

[output_pin probe_enable]
pin: EBBCan:gpio22  # Set to the control pin on your board
value: 0

[probe]
pin: ^!EBBCan:gpio21  # NOTE FOR V2 users: Set this to ^!PC14 to set the low level trigger 
deactivate_on_each_sample: False
x_offset: 0.0
y_offset: 25
z_offset: 0  # Will be overridden when you do a PROBE_CALIBRATE
samples: 2
samples_tolerance: 0.05
samples_tolerance_retries: 3
activate_gcode:
    PROBE_DOWN
    G4 P500
deactivate_gcode:
    PROBE_UP`

looxonline, 2024-06-24:
> @looxonline right, but mine is either always triggered even when deployed, or always open even when stowed.
> 
> Here's my config:
> 
> ```
> gcode:
>     SET_PIN PIN=probe_enable VALUE=1
> 
> [gcode_macro PROBE_UP]
> gcode:
>     SET_PIN PIN=probe_enable VALUE=0
> 
> [output_pin probe_enable]
> pin: EBBCan:gpio22  # Set to the control pin on your board
> value: 0
> 
> [probe]
> pin: ^!EBBCan:gpio21  # NOTE FOR V2 users: Set this to ^!PC14 to set the low level trigger 
> deactivate_on_each_sample: False
> x_offset: 0.0
> y_offset: 25
> z_offset: 0  # Will be overridden when you do a PROBE_CALIBRATE
> samples: 2
> samples_tolerance: 0.05
> samples_tolerance_retries: 3
> activate_gcode:
>     PROBE_DOWN
>     G4 P500
> deactivate_gcode:
>     PROBE_UP 
> ```

Fixed:

[probe]
pin: ^EBBCan:gpio21

looxonline, 2024-06-24:
> @looxonline right, but mine is either always triggered even when deployed, or always open even when stowed.
>
> Here's my config:
>
> ```
> gcode:
>     SET_PIN PIN=probe_enable VALUE=1
>
> [gcode_macro [...]

So you are saying that the triggered state does not change at all when the probe changes between stowed and deployed?

IggyTheWolf, 2024-06-24:
> So you are saying that the triggered state does not change at all when the probe changes between stowed and deployed?

Yes. If I add ! to the probe pin it's always triggered and if I remove it, it's always open. 

I think my probe is defective since the same thing happens when it's plugged into GPIO6.

cruzer619, 2024-06-24:
question.. shouldn't  [bltouch] be used.  even if its a microprobe.  so it knows there is a sensing pin.. which from looking at the support page in biqu..  I just never used [probe] for bltouch type devices.  tap or kclicky yes..  none sensing probe types.  again I could be wrong.. but thats how i've used it with klipper in the past.

cruzer619, 2024-06-24:
This is my config :

[bltouch]
#control_pin: PB1
#sensor_pin: ^EBBCan:gpio21
 sensor_pin: ^EBBCan:gpio24
 control_pin: EBBCan:gpio22
 x_offset: 0
 y_offset: -23
 speed: 20
 samples: 2
 sample_retract_dist: 5.0
# pin_up_reports_not_triggered: True
# pin_up_touch_mode_reports_triggered: True
# probe_with_touch_mode: True

IggyTheWolf, 2024-06-25:
> Fixed:
>
> [probe] pin: ^EBBCan:gpio21

Is this the case for V2 probes as well?

IggyTheWolf, 2024-06-26:
Ok, I'm at a loss here. I purchased another Microprobe and still have the same issues. I then tried a BLTouch connected and configured and the results are the same.

Either the board is faulty or GPIO21 is not working properly. Has anyone gotten a sensor to work on GPIO21?

kgadberry, 2024-06-26:
> Has anyone gotten a sensor to work on GPIO21?

I have a CRTouch working on GPIO21. It's configured to set the pin high (output 5v) instead of the default pulling it to ground.
It was also working when I tested it with a 10k resistor acting as a pull-up resistor.

IggyTheWolf, 2024-06-27:
Ok, I am at a complete loss here. I purchased another EBB SB2209 thinking it could be a bad board, but the same issue persists. So far I have tried the following.

1. Two separate Microprobes and BLtouch. (tested directly to the board and they work)
2. 10K resister on first EBB between 5v and GPIO21
3. A new EBB board

I'm using the X endstop pin so I can't switch to that one. Is there anything else I can do? Is this a bug with the GPIO21 pin?

drowfx, 2024-06-27:
All I can say is that it works for me with a filament sensor switch with the 10k resistor mod. Without the resistor the pin was always reading low, with the resistor the pin reads low when the switch is closed (to GND) and high when the switch is open. I don't have a Microprobe or BLtouch so I can't do any further testing there, sorry.

cruzer619, 2024-06-27:
> Ok, I am at a complete loss here. I purchased another EBB SB2209 thinking it could be a bad board, but the same issue persists. So far I have tried the following.
>
> 1. Two separate Microprobes and [...]

Yes my workaround that I listed in the 2nd post of this thread...  and the picture of where you need to jumpers.

IggyTheWolf, 2024-06-28:
@cruzer619 Thank you! I can't believe I didn't try that after reading this whole thread. That works perfectly! Now I have an extra EBB board and microprobe. (:

bigtalljv, 2024-09-20:
Hi, I also have the same issue.  I am working on an ERCF so I need to more toolhead filament switches.  I already added one to GPIO6 and it’s fine but I need another and the only pin I have is GPIO21 and it doesn’t work.  The electrical talk is a little over my head but I have a Voron Tap and it looks like it’s got three wires, power, ground, and signal? (For my simple brain).  Could I move the signal over to GPIO21 and hope it would work because the Tap is getting power separately?   Would GPIO22 then work “normally “ with a micro switch?  

Thanks

cruzer619, 2024-09-20:
@bigtalljv  you can try to see if tap will work with gpio21 let me know if that works but I have a feeling it won't.  I recently switched to chaotlic lab metal tap.  So I could free up gpio6 for filament switch.  And was going to try and test 10k resistor on gpio21 when I get the ERCF V2.  As others have stated they it working with a filament switch.  Just wish BTT would get the io port fixed.

bigtalljv, 2024-09-21:
> @bigtalljv you can try to see if tap will work with gpio21 let me know if that works but I have a feeling it won't. I recently switched to chaotlic lab metal tap. So I could free up gpio6 for filament switch. And was going to try and test 10k resistor on gpio21 when I get the ERCF V2. As others have stated they got it working with a filament switch. Just wish BTT would get the io port fixed.

Nope, GPIO21 doesn’t work for Tap.  Guess I’m off to the store for a resistor…

serbiansky, 2024-12-07:
I am planing to use Eddy Duo in CAN option and BTT smart filament sensor connected directly to bard. Do I need to solder 10k resistor or in my case it is not required?

cruzer619, 2025-01-30:
Just wanted to post that finally got my ERCF V2 and starting to build it.  But first thing I did was solder on 10k resister and test pins.  And 10k resistor worked like a charm for gpio21.  Thanks you @drowfx   Just have to solder resistor on back of board between pins 2 and 5 like image below.  Will be using gpio21 and gpio6 for toolhead sensors.

[image]

TurboTronix, 2025-05-25:
> Just wanted to post that finally got my ERCF V2 and starting to build it.  But first thing I did was solder on 10k resister and test pins.  And 10k resistor worked like a charm for gpio21.  Thanks [...]

So gpio21 and gpio6 are available to use for filament sensors? What does your config file look like? When i use 21 i get an error that it's already been used.

cruzer619, 2025-05-26:
[image]

akm25-v24, 2025-07-31:
All did not work for me, what I did the old fashion, running the signal wire from the probe to one of the endstop pins of the M8P. Sorry, I don't have the enough electric skills to do soldering and .....