bigtreetech/EBB #90: Unable to read UART EXTRUDER

wookie666, 2023-12-19:
I have an ebb36 connected to a u2c.
All fans and end stops work. 
heater and thermistor works. 

The only thing I can’t get working is the extruder tmc2209. Using the config from the GitHub with PA15 as the pin. I always get unable to read tmc uart. 

I’ve tried setting the uart address to 0. Tried all combinations of fixes online. 

Completely turned off power to the main board and the raspberry pi, including to the canbus boards. Still no luck 

when using the extruder on the main board pins using a tmc2209 driver it all works. The x y and z also work. So it’s just the canbus tmc2209 I can’t get to work. 

Any ideas?

wookie666, 2023-12-19:
Canbus clearly works, fans, heater etc all work
[image]
When using PA15 for the uart_pin, I get no read on it. When using P1.1 which is on the mainboard, with a tmc2209 plugged in, it works fine.
Its just the Extruder on the canbus board i cannot get to work.

`[tmc2209 extruder]
uart_pin: EBBCan: PA15 #P1.1
interpolate: true
run_current: 0.850
hold_current: 0.100
sense_resistor: 0.110`

primate25, 2024-02-24:
Did you resolve the issue?