bigtreetech/EBB #91: Can the USB port be used to support a nozzle camera?

cray12399, 2023-12-20:
I have a BTT EBB SB2209 (RP2040). I am thinking of adding a 3DO nozzle cam to my setup. However, I do not want to purchase it unless I can install it to my board because I do not want to compromise my single cable umbilical setup.

Can I install a nozzle cam to the USB port on the board, or is the USB port used exclusively for flashing?

Alternatively, if I cannot use the USB port, is there another way to connect a nozzle cam to my board that will allow me to still use one cable?

mutlubasdas, 2024-04-07:
Hi there! Did you find a way for this?

looxonline, 2024-04-28:
Not just yet but we will soon have a prouct that makes this possible.

mutlubasdas, 2024-04-28:
That would be a good selling product imo.
I'll be following this.