bigtreetech/EBB #93: original AKA v1.1 AKA main_body_EBB_SB.stl has been deleted from repo thus breaking all known community PCB cover mods

kevinhamon, 2023-12-28:
@bigtreetech The commit https://github.com/bigtreetech/EBB/commit/35baceb5cac23f437ba8bebc3f8c5f33144fc09a breaks all known community PCB door mods because of a puzzling decision to add a chamfer to the main body where the door meets the main body. The chamfer is mated up in the corresponding v1.2 PCB cover, but since you've deleted the original main body file, no community mods will work with the only main body currently available in the repo. 

For anyone who needs the original file (that will work with community mods for EBB2240/2209), you can get it here: https://github.com/bigtreetech/EBB/blob/c4c6e7c620f44e13a288c115c6ca01aedfa1239f/EBB%20SB2240_2209%20CAN/Custom%20Printed%20Parts/main_body_EBB_SB.stl

@Greecher has posted files that worked for me [here](https://github.com/bigtreetech/EBB/issues/65). You will want the main_body_EBB Fix.3mf version. 

The community will surely catch up but, at least for the short term,  someone from btt needs to resurrect the original main_body_EBB_SB.stl and either version it as as v1.1 or just leave it as is and move it to a superseded_parts folder along with branching/tagging the repo to track changes like the rest of the Voron repos do. 

At this point, I would suggest to rename the file as v1.1 as you've now added confusion by starting a new versioning scheme mid lifecycle with the addition of v1.1 and v1.2 naming on other files. Should have started using versions from the start, or better yet, followed Voron's scheme - it makes a lot more sense. --- Sorry, I now see that you can't branch / tag because you have a monorepo for CAN/EBB based projects. Oops, that was an unfortunate decision.

`<rant>`
This will be the **3rd time** printing this set of parts for my 2.4 build because there are many different variations of the main body and covers floating around that don't work together. I expect this for experimental community mods or even when I have a unique set of features I am trying to combine.. but neither is really the case here.. I should be able to come to this repo to get everything needed for a base build that won't burn up the boards. 

The heat issues on these boards are well documented. Still @bigtreetech only supplies an unvented solid cover in this repo. **_Please_** at least supply a vented cover, and/or something that accepts a 3010 or 3007 for cooling. If a vented cover would have been available here, I would have started with that, and finished with the tool head days ago. If my specific needs warranted,I would have upgraded to something more actively cooled after completing a base build. 
`</rant>`

Rowan2233, 2024-03-18:
Hey sir, do you mean the Main_Body STL in the customer print part is design for the community Cable_Cover_For PCB? We delete the customer print part file cause we got a lot complain form our user, they said the Main_Body STL in the customer print part file isn't the original design from voron team, it lead people to worong installed. And can not work with our Cable_Cover_For_PCB we push before since the it is design for the Voron team original Main Body.

Right now, we have design a new Main_Body_EBB, it could better fit our EBB SB CAN series, but it need be work with Cable_Cover_EBB;  However you can use Cable_Cover_For_PCB_V1.1 if you still use the original Main_Body  STL from Voron team.

AhmedSoliman, 2024-04-07:
@Rowan2233 The STL file that was removed in https://github.com/Rowan2233/EBB/commit/715629e564b63ccb8f705a548d57c549e4291d9d was the correct one that matches the motor plate from SB's repo. The new one doesn't fit the motor plate anymore.