bigtreetech/EBB #94: EBB SB2209 RP2040 : CR/Bltouch always open

Ixian, 2024-01-19:
Same as this ticket: https://github.com/bigtreetech/EBB/issues/88 which the OP closed because they found a workaround using a gpio from another port. That isn't an actual fix though and BTT needs to know that it is still shipping broken SB2209 RP2040 boards. The CR/BLTouch port does not work because GPIO21 either has a broken pullup or a weak one. As a result the signal state never changes because of the float (which a pullup would normalize). 

You can tap the proximity or endstop connectors and use their GPIO instead, but that then blocks you from using said port for anything else. This needs to be fixed.

bugalaws, 2024-02-25:
I have the same issue, do you solve this ?

cruzer619, 2024-02-27:
> I have the same issue, do you solve this ?

@bugalaws did you try my workaround to get it working?  #88

bugalaws, 2024-02-27:
> > I have the same issue, do you solve this ?
> 
> @bugalaws did you try my workaround to get it working? #88

Yes, i try this but doesnt work for me. No work the bltouch niether the rgb led… for now i put a separated cable conected at the printer board to drive the bltouch and the leds… mas  buy a new btt board because i think my board have some problem…