bigtreetech/EBB #97: BTT SKR3  EZ + EBB36 = not communicating

BazlTech, 2024-02-17:
I've followed the guides in this git repo, I've followed other guides in blog posts elsewhere.  Nothing I try actually makes this work.

Setup:
Raspi 4B running Klipper
BTT SKR3 EZ connected to Raspi via USB
BTT EBB 36

I've reflashed firmware to the SKR3 just to make sure I had the settings correct.  I've flashed CANBoot to the EBB36.  I've flashed Klipper to the EBB36, but nothing works.

For what it's worth, I've never been able to get this command to work: 
~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0

sntlewis, 2024-02-17:
two things (neither may apply to your case).  1) power to the EBB36?

2) While the PAD7 I run supposedly supports CANBUS, I never could get it to recognize the EBB36.  Added U2C and all was straight forward.

Like I said, neither may be helpful..

BazlTech, 2024-02-20:
I was able to determine the proper sequence of events to make this work.  BTT should really do a better job with their user manuals.  What made my issue more complicated was I already had a working system before integrating the EBB36.  Things may have been more obvious if I had started completed from scratch with all of the equipment at once.

Here is the short and sweet of it:

1. Flash the SKR 3 EZ with Klipper, but select the USB to CAN option (I think the CANBUS pins are PB8/PB9?).  I used the following method for this:  Rename klipper.bin to firmware.bin.  Copy file to microSD card. Insert into SKR.  Power up for 5 seconds, power down and remove SD card.
3. Flash the EBB36 with Klipper configured to run over CAN (PB0/PB1) using the USB-C connection (you'll need to use the jumper to select USB power).  **DO NOT DO THIS WITH THE SKR3 POWERED UP** if you have your EBB36 already connected to the SKR.
4. Connect CANBUS wiring from EBB36 to SKR 3 EZ.  Swap the jumper over to the 120R resistor position.
5. Power up the SKR.
6. Using your Pi (that is connected via USB to the SKR), you'll need to use the python command to view your CANBUS devices and grab their IDs.
7. Edit your printer.cfg file's MCU sections (main one for the SKR, the secondary "mcu EBBCan" section for the other CANBUS ID)
8. Restart firmware
9. PROFIT

closed 2024-02-20 by BazlTech