bigtreetech/EBB #103: EBB2240  with PT1000 clarification

donmahallem, 2024-02-21:
I do have a question regarding the **EBB2240** boards. I do use a Rapido V2 hotend with a **TWO wire PT1000** Thermistor. While looking for something else in the [schematics](https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2240_2209%20CAN/SB2240/Hardware/BIGTREETECH%20EBB%20SB2240%20CAN%20%20V1.0_SCH.pdf) I did notice that the board specifies a 4300R resistor while the Klipper [docs](https://www.klipper3d.org/Config_Reference.html#directly-connected-pt1000-sensor) specifies a 4700 Pullup resistor. Am I correct with my assumption that I do have to replace the 4700 default with the 4300? (I did set the dip switches to the PT1000 position before). I just want to make sure before I go ahead as this would mean a significantly higher temperature

closed 2024-02-21 by donmahallem

Karl-Barx, 2025-02-13:
Did you ever figure this out I'm having a hell or a time with mine

donmahallem, 2025-02-13:
Can't look up the dip switches currently but I have this config running for several months unchanged:
```sensor_type:MAX31865
sensor_pin: EBBCan: PA4
spi_bus: spi1
rtd_nominal_r: 1000
rtd_reference_r: 4300
rtd_num_of_wires: 2```

bigtreetech, 2025-02-13:
The EBB2240 has a MAX. The user is running an EBB USB which does not. His config will be different.