bigtreetech/EBB #112: TACH on SB22** Boards does not work.

M3HNGRY, 2024-03-22:
The tach do no work on the SB2240 with 9200rpm 4wire 5015 fan. My slider works but rpm/tach doesn't change. 😦https://cdn.discordapp.com/attachments/1015253264935157861/1220590923080142868/image.png?ex=660f7ee1&is=65fd09e1&hm=6860d3c02bde2fa892c3b3df819aaf957b78939b90a4d9a32e4929e7534ed763&

RED = 24v, Black is Ground, White = Tach and Blue PWM...
klipper v0.12.0-131-gd9043345 

https://cdn.discordapp.com/attachments/1015253264935157861/1220592102321291334/image.png?ex=660f7ffa&is=65fd0afa&hm=49c510f1f8b60e8581e75160a03108e246b9179648f9646a6f14f5ac32774250&

MattiaFontanaErrepinet, 2024-03-29:
Same problem on SB2209

chhe-01, 2024-04-11:
Hello, I think I might have the same problem. I use the SB2209 RP2040

B0rax, 2024-04-17:
For reference, this config works for me:
```
[fan]
pin: EBBCan: PB14
tachometer_pin: EBBCan: PB15
kick_start_time: 0.2
hardware_pwm: true
off_below: 0.3
```

christianhelbig, 2024-05-25:
> Hello, I had the same problem. Did you set the jumper on the V_4W-FAN? When I set it, the tacho worked for me.

Thank you so much! I had the same problem and this actually fixed it!