bigtreetech/EBB #113: EBB SB2209 with filament sensor

jbutler610, 2024-04-05:
I have a question on whether or not I can connect a filament sensor to extra pins on my board. 

I am using the tap mod so those pins and the x-limit switch is taken. I saw another post with a 2209 where someone said you could use the proximity switch pins if that is not in use, obviously signal and ground only. I just wanted to clarify that it would work (since I am using the tap) and not interfere with anything else or blow my board up. 

That would save a ton of time wiring if possible.
Thanks

Delfairen, 2024-07-11:
Did you get an answer to this. It does look like the filament sensor can be connected to GND and GPIO6 of the proximity connector as the Tap goes to Vin, GND and GPIO22 but wanted to confirm you had done this and it worked.

m3cinc, 2024-07-12:
I have 2 filament sensors connected and in operation with my voron 2.4 ercf
: the extruder sensor uses GND/PB63 and the toolhead sensor GND/PC13.
The CNC TAP uses PB5/GND/+5V.
I could still add a filament runout using PB8/GND or PB7/GND.
Of course, I am using sensorless homing with virtual endstop.
All works with KLIPPAIN/Happy Hare.

Marc

cruzer619, 2024-09-23:
Look at #86 will help you.

Cracoukas, 2025-01-12:
> Did you get an answer to this. It does look like the filament sensor can be connected to GND and GPIO6 of the proximity connector as the Tap goes to Vin, GND and GPIO22 but wanted to confirm you had done this and it worked.

I first used GPIO21 but for some unknown reason it was very unstable. GPIO24 is Ok (of course woth sensorless homing)