bigtreetech/EBB #114: SB2209/SB0000 Fans not working

brothereww, 2024-04-12:
After installing Klipper v12 everything thats connected to the SB0000 stopped working (Part/Hotend Fan and a Knomi Display).
The Hotend itself is Heating and the Thermistors are working too so i dont think its an issue on the SB2209 my guess is the SB0000 is not getting any current but im not really sure how/where to measure. 
- I didnt change any wiring everything is as before the Update where everything worked just fine
- I didnt change anything in the cfg files (I did try to reverse the PA1 and PA0 in the cfg but it didnt work)
- I already replaced the SB0000 Board

Im running a Manta M8P Board with a CB1 and the SB2209 in USB Can Bridge Mode.
Everything else is working fine I also dont get any error reports.

dmradford, 2024-04-13:
I'm having the exact same issue. Manta M8P in usb-can bridge mode. Everything connected directly to the SB2209 (RP2040) board works correctly. I only have the standard 2 fans and RBG LED's connected to the SB0000, but I'm getting nothing. I assume the pinout is just wrong, but I've tried everything I can find.

brothereww, 2024-04-14:
I also have no clue what to do at this point but sure know as soon as it runs again im ditching the mysterybox that the voron is super pissed with this thing i have more hours in tinkering with this thing than it has print hours

anatolygor, 2024-06-04:
Same here. Did anyone resolve this issue?

brothereww, 2024-06-07:
I bought a new SB2209 and it worked again. I dont know what the issue with the old one was im sorry

qw3rtn, 2024-10-10:
Any news on this topic?

Everything works except for the fans on the SB0000 :/
SB2209 running in can bus mode.

marceneirojovem, 2024-11-20:
Same here with Neopixel led and Knomi 2

stokedgsxr, 2025-05-13:
> Any news on this topic?
> 
> Everything works except for the fans on the SB0000 :/ SB2209 (RP2040) running in can bus mode.

Did you get this resolved?

qw3rtn, 2025-05-13:
> Did you get this resolved?

Not really. I bought another SB2209.
I guess I misaligned the pin header when putting all together and fried something on the SB2209 :(